RudolphRiedel/FT800-FT813 #2: Drawing basic shapes

ttlappalainen, 2018-03-24:
Hi,

I found your library and FT813 powered displays very interesting and ordered RVT70UQFNWC00 for testing. I am developing system where I currently use RA8875 controller, but it start to lag on my project specially due to bad font quality. 

With a quick look for FT81x Series Programmers Guide I started to wonder is there any way to draw framed rectangles or circles without fill? As far as I found RECTS or POINTS will allways fill the area, I would also need other basic shapes like arc.

RudolphRiedel, 2018-03-30:
I am afraid that the FT8xx do not offer much in this direction.
At least not directly.
Main focus is on objects, less on graphics primitives.

The available graphics primitives are limited to Bitmaps, Points, Lines, Line-Strips, Edge-Strips and Rectangles.
So no Arcs.

For frames I just put a Rect and then annother smaller Rect on top of it with the background-color.
Another way would to use four lines or even better a line-strip.
Using a line-strip requires four coordinates, using two rectangles also reduires four rectangles.
Drawing a linestrip does not require to change the color between coordinates though.

ttlappalainen, 2018-03-30:
On some samples I think I saw they did arcs with gameduino port. Also found some nice gauges - much better than build in. I wonder how they have done those. I also tried to find way to add custom vector objects, but it seem to support only bitmaps. 

Do you know better display for own graphics than RA8875. It should have good fonts to large size. Also would like to have antialiasing handled internally.

RudolphRiedel, 2018-03-30:
I just searched for gameduino and arc and came across this:
http://gameduino2.proboards.com/thread/106/drawing-arc
So, two overlappings points for the circle and an edge-strip to cut the segment.
But the recommendation there really is to use a picture.

The fancier gauges are done with pictures as well.
One for the background and one for the hand.
You can very easily rotate images with the FT8xx including anti-aliasing.
It cames at almost no cost and you can rotate by a fraction of a degree.
 
And regarding "better than RA8875" - you are asking the wrong guy. :-)

ttlappalainen, 2018-03-30:
The bitmaps would be OK, but if I need several different bitmaps, they does not fit to RAM_G, which I understood to be 1MB. It would help, if display would save images as JPEG on RAM_G, but if I understood right, it will change them to bitmaps.  Also it is not possible to draw gauge with different range colors. Or is it possible draw a arc as in your link and apply that to display with transparent bg so that previour arc will not be cleaned away?

RudolphRiedel, 2018-03-30:
As for the size of RAM_G, a 800x480 pixel full color picture uses 750k of RAM.
So for a single screen there should be no problem.

Then you actually can stores images in RAM_G as JPEG by using CMD_MEDIAFIFO to setup the memory the raw data is stored at as the region FT8_cmd_loadimage() will load the data from when using FT8_OPT_MEDIAFIFO.

Then you could cheat it a bit by not using a full image, in case of a circle you could use a quarter of that and display it four times with the extra three instances rotated of course.
As well as streching the image would be possible thus allowing different sized images from a single tile.

Colors and Alpha-Blending is possible but I have not touched that so far.

Next helpfull option will be the external FLASH the BT815 will be able to use, but that has only been announced so far.

As for this discussion, I feel this is not the right place, hardly anyone else will see it and contribute.
If you like to continue, please ask here: https://www.mikrocontroller.net/topic/395608#new
Yes, the forum is German, just ignore that. :-)
Posting there will definately increase the change of others adding input to it.

And then you just can start playing with the RVT70UQFNWC00 you ordered and see where this ends. :-)

ttlappalainen, 2018-03-31:
I'll test. Thanks a lot for all info.

closed 2018-03-31 by ttlappalainen

dstulken, 2020-10-16:
> For frames I just put a Rect and then annother smaller Rect on top of it with the background-color.

For what it's worth, the same can be done with button widgets (no text, no tag), using the "EVE_OPT_FLAT" option on both, to give frames with nice rounded corners!

RudolphRiedel, 2020-10-16:
No, that is a rather ineffcient way to draw rectangles.

But lets have a close look with EVE Screen Editor.
I wrote frames and meant not filled rectangles.
Here is a flat button and a frame:
[image]

Now the button needs 5 32-bit values on the host side if you do not change the color.

CMD_BUTTON(183, 67, 120, 36, 27, OPT_FLAT, "")

And EVE translates the command to this for the display-list:
SAVE_CONTEXT()
VERTEX_FORMAT(2)
LINE_WIDTH(60)
BEGIN(RECTS)
COLOR_RGB(0, 56, 112)
VERTEX2F(747, 283)
VERTEX2F(1197, 397)
BEGIN(BITMAPS)
RESTORE_CONTEXT()
SAVE_CONTEXT()
VERTEX_FORMAT(2)
BITMAP_HANDLE(27)
RESTORE_CONTEXT()

That are 13 32 bit words in the display-list.
A good portion of this is just utter nonsense.
But it gets worse, when you add annother button the exact same sequence is added again.

Drawing a frame like in the picture with rounded edges looks like this:
BEGIN(RECTS)
LINE_WIDTH(164)
COLOR_RGB(255, 0, 0)
VERTEX2F(2200, 2200)
VERTEX2F(5500,3500)
COLOR_RGB(0, 0, 0)
VERTEX2F(2250, 2250)
VERTEX2F(5450, 3450)
END()

That is 9 32 bit words on the host side and for the display-list.
And it already includes the color change.

Now we need three frames:
[image]

BEGIN(RECTS)
LINE_WIDTH(164)
COLOR_RGB(255, 0, 0)
VERTEX2F(2200, 2200)
VERTEX2F(5500,3500)
VERTEX2F(6000, 2000)
VERTEX2F(8100, 2700)
VERTEX2F(6200, 3200)
VERTEX2F(8100, 3700)
COLOR_RGB(0, 0, 0)
VERTEX2F(2250, 2250)
VERTEX2F(5450, 3450)
VERTEX2F(6050, 2050)
VERTEX2F(8050, 2650)
VERTEX2F(6300, 3300)
VERTEX2F(8000, 3600)
END()

dstulken, 2020-10-16:
Whoops - Can you tell I'm new to these boards?  :D
I didn't realize that the rectangle primitive could draw rounded corners natively!
Thanks for correcting me.

RudolphRiedel, 2020-10-16:
That is not an obvious feature. :-)
But it is documented in the programming guide, 2.10.4 "Drawing Pattern" of the BT81x guide has examples that show this.