RudolphRiedel/FT800-FT813 #10: New native ESP32 SPI with DMA and improved Arduino SPI

davidjade, 2020-05-15:
This is not really an issue per-se, but I recently re-worked your code and I thought I would share my improvements back. It is a lot of fundamental changes so rather than submit a huge pull request that you may not want, I'll just point you to it here:

https://github.com/davidjade/FT800-FT813/tree/4.x

My main motivation for doing this was to find a way to enable the LittlevGL GUI library to interface with FT81x boards, which mostly required a lot of performance work. I think I have been successful in that as I now have a striped-down driver for LittlevGL (not include here) that works. My new LittlevGL display driver can do full screen updates at about 4-5 FPS using 2-wire SPI DMA mode and should be up to 4x faster with Quad SPI (which I am working on next). This new ESP32 SPI DMA code is stable for me at 30Mhz on a breadboard, which is significantly faster than the Arduino demo is capable of.

Some stats on my performance work (these are the numbers displayed in the demo):

Original FT81x Arduino demo - stable for me at 16Mhz
- time1: 1586us
- time2: 94us
	
New FT81x-Arduino - stable for me at 16Mhz
- time1: 191us
- time2: 45us

New FT81x-ESP32 - stable for me at 30Mhz
- time1: 147us
- time2: 27us

I'd be happy to make this a pull request if you want but please take a look first. I only updated the PlatformIO Arduino demo project and added a new native ESP32 PlatformIO demo project and have left the main EVE code files in the project root unchanged for now. I most certainly broke the existing DMA code for boards I don't have and also have not tested on any other Arduino boards as I only have ESP32 boards. 

If this does not interest you that's Ok. Or if you just want to take some ideas from this on your own, that's fine too. I just wanted to share back as your code was very helpful to me.

RudolphRiedel, 2020-05-15:
First of all, thanks for sharing!
I am always interested in expanding the supported platforms!

So I pulled the CFAF320240F kit with the Arduino Uno clone from the shelve and put the latest version of my Arduino test code on it. Yes, that is the same as in the Github.
time1 : 704µs
time2: 48µs

Now I tried to compile your EVE_Test_Arduino_PlatformIO project.
Okay, the platformio.ini had no section for the uno, so I added it.
Now it tries to compile but it can't since the SPI class for the Uno does not have
a method "transferBytes" and a couple of extra errors are popping up.

Okay, checking the code.

 SPI.setFrequency(16000000);
 TFT_init();

Two problems with this, this first is that the SPI class does not have a method "setFrequency".
The second is that this violates the requirement for EVE that during setup the SPI must be slower than 11MHz.

So for ESP32 this should look like this:

 SPI.setFrequency(10000000);
 TFT_init();
 SPI.setFrequency(16000000);

In EVE_target.h you replaced several one-line functions for the SPI-transfers with a combined function.

Well, looks like you got this to work, great.
But this could be done much simpler and faster as everything to support DMA already is in place.

Check out the section in EVE_target.h right under the licence:

xxxx
  For the SPI transfers single 8-Bit transfers are used with busy-wait for completion.
  While this is okay for AVRs that run at 16MHz with the SPI at 8 MHz and therefore do one transfer in 16 clock-cycles,
  this is wasteful for any 32 bit controller even at higher SPI speeds.

  Check out the section for SAMC21E18A as it has code to transparently add DMA.

  If the define "EVE_DMA" is set the spi_transmit_async() is changed at compile time to write in a buffer instead directly to SPI.
  EVE_init() calls EVE_init_dma() which sets up the DMA channel and enables an IRQ for end of DMA.
  EVE_start_cmd_burst() resets the DMA buffer instead of transferring the first bytes by SPI.
  EVE_end_cmd_burst() just calls EVE_start_dma_transfer() which triggers the transfer of the SPI buffer by DMA.
  EVE_cmd_start() just instantly returns if there is an active DMA transfer.
  EVE_busy() does nothing but to report that EVE is busy if there is an active DMA transfer.
  At the end of the DMA transfer an IRQ is executed which clears the DMA active state, calls EVE_cs_clear() and EVE_cmd_start().
xxxx

So with "#define EVE_DMA" a couple of things are changed.
A buffer is expected to be setup together with a pointer for this buffer.
The functions EVE_init_dma() and EVE_start_dma_transfer() are expected to exist and for these I have the module EVE_target.c.

Then spi_transmit_async() is changed to only write to the buffer:

static inline void spi_transmit_async(uint8_t data)
{
 #if defined (EVE_DMA)
  EVE_dma_buffer[EVE_dma_buffer_index++] = data;
 #else
  SPI.write(data);
 #endif
}

The native ATSAMD51/E51 version with DMA running at a core clock of 120MHz and a SPI speed of 15 MHz is pretty fast:
time1 : 35µs
time2: 8µs

So the ESP32 running with a core clock of 180MHz should be doing a little better than that.

Now I can not really setup the ESP32 functions for you and I have no way of testing these.
But I guess since it is Arduino, EVE_init_dma() should be mostly empty.

void EVE_init_dma(void)
{

 // setup an end of DMA transfer interrupt somehow?
}

And EVE_start_dma_transfer() may look like this:

void EVE_start_dma_transfer(void)
{
 EVE_cs_set();
 SPI.transferBytes(EVE_dma_buffer, NULL, EVE_dma_buffer_index); // ?? probably the wrong function, we need a non-blocking version here
 EVE_dma_busy = 42;
}

Finally an end of DMA function could look like this:

void EVE_DMA_end(void)
{
 if( test somehow that DMA is done)
 {
  EVE_dma_busy = 0;
  EVE_cs_clear();
  EVE_cmd_start(); /* order the command co-processor to start processing its FIFO queue but do not wait for completion */ 
 }
}

This is of course the polling version of this as I have no idea how to add an interrupt function for the ESP32, so this could be called in the loop like this:

if(EVE_dma_busy)
{
 EVE_DMA_end();
}
else
{
 // talk to EVE again
}

davidjade, 2020-05-15:
Thanks for the feedback. I'm happy to share this even if it is not something that you can use.

As I mentioned, my main reason for doing this was to create a FT81x driver for the LittlevGL GUI framework. That framework controls all rendering via pushing screen updates as bitmaps to regions of a TFT. Yes, this defeats the purpose of using something like the FT81x since you don't use any of it's GPU features but those were my needs. In this driver I created, I did this by creating a DL with one full screen bitmap. To update that bitmap I transfer RGB565 pixel data directly to the RAMG region of the bitmap in the FT81x display via the memory mapped SPI protocol. Doing this I can refresh the entire 800x480 16-bit screen somewhere between 4-5 frame per second. Each screen refresh is 6,144,000 bits (768k bytes) so that's roughly the 30Mhz SPI transfer speed - i.e. the full bandwidth of the FT81x. This was not even close to possible using any other method on the ESP32 without fully leveraging the features of the ESP32's DMA API. It is also much, much faster than using any of the GPU features for bitmaps on the FT81x as well. Working with such large uncompressed bitmaps also implied some design choices that I made.

A few more notes and thoughts about the choices I made:

Yes, I was worried that some Arduino frameworks might not have all of the same SPI functions but I think it should be possible to fall back to using the manual loops (i.e. creating a local version of writeBytes) for those that don't allow transferring blocks at a time. I did not write this code though as I don't have other platform to test on or even know which ones would need it. The demo code I checked in was very much only verified on an ESP32 and I suspected it would only run there at this point.

Interesting about the clock speed throttling at start up, I missed that part of the spec but it is interesting that it has always initialed Ok for me at 30Mhz (32Mhz actually). Probably not the safest choice though.

Lastly, I did see the DMA code you wrote but if you look at the new demo I created for the ESP32 using native SPI DMA, well, the ESP32 SPI DMA is a different beast and the existing DMA buffering was not flexible enough, especially when using such large buffers as I needed. The Arduino changes started from this new DMA code that I back ported to the general case for Arduino, just because I could - I actually have no use for Arduino code but I thought it might be useful since it seemed to have advantages even without using DMA.

ESP32 DMA notes:

1. For DMA on the ESP32 using the platform's spi_master really works best to let it control the CS line itself, so that was one part of the problem. This means that the CS toggles during the DMA transactions - not in any of my code.
2. For ESP32 DMA transfers of large blocks you really need to have the data in one block and the memory mapped SPI address bits as part of the transaction data structure. ESP32 handles this perfectly (as well as command bits, but I don't currently use that). This keeps you from having to duplicate or re-buffer addresses and data into one continuous block. I did not want to do that as in my driver, I sometimes send 786K of bitmap data at once. There was no good way to take an already formed bitmap of that size and then prepend the address bits, commands, etc... 
3. I also had the need in my LittlevGL code to create my own windowing method for partial screen updates and this requires many individual partial screen writes dissected from one block of bitmap data. Again, to avoid re-buffering data I used the ESP32 SPI transaction structure features to send address bits and this made it much more efficient than buffering address bits and data together.
4. My new FT81x code both uses the transparent SPI buffering technique and allows for direct queuing of DMA transactions so that when you have large blocks of data you don't have to double-buffer to use DMA. 
5. DMA are queued SPI transactions and need to be retrieved to clear internal memory, so a little more housekeeping is needed but this is handled transparent in the native part of the ESP32 DMA code and isolated from the FT81x code.
6. It is also possible to queue or stream multiple DMA transfers and let the ESP32 SPI driver handle all the housekeeping (not enabled right now but it could be easily done). I could enable this now and the FT81x code would benefit with no changes required (if fact it would not even know).
7. There are also some restrictions on what type of memory, alignment, and how big the chunks can be for DMA transfers and this required a little more control. This required some cooperation with the FT81x code (i.e. chunking).
8. Lastly, the pushes the decision over whether to transmit byte-by-byte or by complete blocks when the APIs exist down to the target layer so that frameworks that can send complete blocks benefit without the FT81x code changing.

So I strove to follow best practices for ESP32 as far as I knew, which lead to the differences between your and my methods for handling DMA. However, I also tried to write my DMA code in such a way that I think it could be adapted to other platforms and frameworks. The generic SPI buffer and handing off of SPI transactions was my attempt at that and it also has benefits for platforms that don't have DMA transfers while still enabling platforms that have rich DMA support to leverage advanced features. This also had the side benefit of reducing code  [... cut ...]
So that's my perspective after realizing that the existing DMA code was not really a good fit for the ESP32 and my needs. It's just that my needs exceeded what was possible with your DMA framework.

As for example demo speed, I'm not sure where the ESP32 DMA example fits in the scheme of things compared to your other demos but the raw bitmap writing that I can do seems to point towards getting pretty close the the FT81x spec with my code. I will say that currently, my DMA code does wait for transaction to clear in some cases so that may be causing some demo times to include those wait times.

Anyway, thanks for the feedback and taking a look at it.

RudolphRiedel, 2020-05-15:
Yes, I inored the LittlevGL bits as this really defeats the purpose of EVE. :-)
A simpler display might be a better match for this aplication.

Anyways, just looking at your application I have an idea to boost it with DMA
without "breaking" everything else.

You changed everything now to use DMA but you are not really using most of it if all you really do is setup EVE and put up a simple display list once to display one image.

I am using the DMA to write to the command FIFO in order to create the display
list using the command co-processor.
And the FIFO only is 4k.

You need to write a huge chunk of data directly to GFX-memory and can do so without truncating it into 4k pieces.

Have a look at this from EVE_commands.c:

void EVE_memWrite_flash_buffer(uint32_t ftAddress, const uint8_t *data, uint16_t len)
{
 uint16_t count;

 EVE_cs_set();
 spi_transmit((uint8_t)(ftAddress >> 16) | MEM_WRITE);
 spi_transmit((uint8_t)(ftAddress >> 8));
 spi_transmit((uint8_t)(ftAddress));

 len = (len + 3)&(~3);

 for(count=0;count<len;count++)
 {
  spi_transmit(fetch_flash_byte(data+count));
 }

 EVE_cs_clear();
}

This function transfers chunks of data directly to EVEs memory.
Okay, yes, looking at it now I should update it to use 32 bits for the length of the buffer.

If all you need is to transfer a raw image to memory you could just make a copy of this function and rewrite it to use DMA.
Everything else really can stay the same.

This shows a flaw in EVEs design for operations like this though, and I do not mean the strange byte-order, the address is only 3 bytes long.
So using a nice aligned buffer of uint32_t does not work directly.
However, this could write a dummy-byte after the address, the only issue with this is that you could not use address 0x000000 anymore for your data.

Something like this:

uint8_t prepare_DMA_buffer(uint32_t ftAddress, const uint32_t *buffer)
{
 if(ftAddress > 0)
 {
  ftAdress -= 1;

  buffer[0] = ((uint8_t)(ftAddress >> 16) | MEM_WRITE))
           + (ftAddress & 0x00ff00)
           + ((ftAddress & 0x0000ff) << 16); 
  return 1;
 }
 else
 {
  return 0;
 }
}

So the data starts at buffer[1].

fill_buffer(buffer);
if(dma_not_busy) send_buffer_over_SPI_by_DMA(buffer);

Yes, no idea if this actually works, you get the idea. :-)
Using two buffers would be an obvious optimisation for this, start filling the next while one is transferred.

If you still need to interact with EVE after this, for example to read the raw coordinates for the touch events, you just need to make sure that no DMA transfer is in progress.

>For DMA on the ESP32 using the platform's spi_master really works best to let it control the CS line >itself, so that was one part of the problem. This means that the CS toggles during the DMA >transactions - not in any of my code.

My DMA code for the ATSAM is not using any library and at least the SERCOM units of the ATSAM I am using can not automatically handle CS transitions for a buffer.
If I would switch automatic CS handling on it would toggle CS for every byte.

Sorry if this is a bit rough but we had a Power Failure when I was about to send the first version of this. The first Power Failure in years but it had to happen when I was about hit the send button...

davidjade, 2020-05-15:
I originally thought I would use more of the EVE features and I still may. One interesting idea is that since LittlevGL just writes to a bitmap, it doesn't have to be a full screen bitmap. It could be just a smaller region. That would allow a mix of both EVE graphics and LittlevGL. It could even be that there are several LittlevGL regions - like split screen, etc... I don't know if I will do this but it's an interesting option. I originally choose EVE displays for two reasons though: First, larger high resolutions capacitance screens and 2nd, the co-processor. In the end the first reason was the more important reason as using LittlevGL for a complex application was far, far easiest than using the co-processor for a complex GUI. Perhaps I will find just as good of a pre-built large screen package down the road but I have several EVE displays so I'll just use them for now, even if I don't use all the features they offer.

That's absolutely true that I am not using much of the EVE features and thus all the DMA work doesn't have much impact. I did it though as part of a broader effort to fully integrate DMA on the native ESP32 platform in case I ever wanted to just use EVE features exclusively. That it also benefited Arduino apps was a bonus (since the ESP32 Arduino supports more efficient transferBytes SPI transfers). Plus the code is smaller in size overall now, both as written and compiled.

> If all you need is to transfer a raw image to memory you could just make a copy of this function and rewrite it to use DMA. Everything else really can stay the same.

Except, on the ESP32 I still need to let the driver manage the CS line for an entire transaction and then all the original code toggles the CS line has to be changed too then. The ESP32 docs specifically call out the dangers of mixing different methods of SPI access methods on a bus. Using a mix of both you may have to tear down the bus config to switch back and forth (there appears to be no way to change the bus config on the fly, which is why their Arduino framework is so slow I think - it appears to do this). And then you have to have the proper code anyway to wait for queue completion too even for single transfers the second you use DMA. This is because they use FreeRTOS queues internally and you really have to keep that internal queue in sync. It made sense to just go with DMA all the way and that, as I pointed out, required larger changes due the the specifics of the ESP32 SPI DMA API.

> Something like this:
> uint8_t prepare_DMA_buffer(uint32_t ftAddress, const uint32_t *buffer)

Unfortunately, that would not integrate into LittlevGL, at least not easily. LvGL manages a rotation of buffers (it will do partial and parallel screen updates, fairly efficiently). I have no way to "prepare" a buffer for it to use and I really don't want to duplicate buffers just to prepend an address. Furthermore, LvGL may decide to update a smaller rectangle of the screen. In those cases, each line of the bitmap has to be written to a different region of the EVE memory. Basically LvGL hands me bitmap data and a rectangle to paint it. So I have to sent addresses for each transfer but I don't have to duplicate any buffers on the ESP32 because it can handle the separation of address and data in its SPI transfer API but then you have to manage all the other SPI features to do it that way. 

btw, I thought about (briefly) using the co-processor to manage a list of bitmaps for each rectangle update but then quickly decided against that and choose the simpler one large bitmap and managing writing sections directly myself. It works quite well since I seem to be able to hit near the full SPI bus speed for updates. Again these are concessions to wanting to use LvGL and I suspect that most people that own an EVE aren't looking for that.

What I wish the EVE had was a screen update method that allowed for "windowed" updates like some other dumb TFT do. It doesn't so I had to roll my own. I can see why they don't though - they likely didn't imagine layering another GUI library on top such as LvGL.

> If I would switch automatic CS handling on it would toggle CS for every byte.

One issue is that the FT81x uses the status of the CS line to determine the end of memory mapped transfers. So it has to be held constant for the destination address and all the data that follows. So they really need to be part of one SPI transaction. On the ESP32, this means letting it manage the CS line for all transfers or you get into a weird state of trying to co-manage and how can you tell when it is safe to use it outside of DMA. It quickly gets more complicated than what I have done.

Sorry if I seem stuck on my DMA method but it is really working quite well for me - even if I drop back down into pure Arduino mode. If I had access to other board types I think I could ensure that it works everywhere with my methods and a few tweaks, but I don't right now unfortunately so I may have to just leave it at that for now.

RudolphRiedel, 2020-05-15:
Well, now as interesting as this is, this definately is way off scope for EVE.
If it works out for you, great!

But if the numbers you posted above are correct, this is way slower than it should be running on a 240MHz 32 bit controller.
And that you have a difference if you run the SPI at 16MHz or 30MHz shows that this is not running really efficent on DMA as it is.
When I slow down my SPI on my 120MHz ATSAME51 from 15MHz to 6Mhz the time1 merely goes up from 35µs to 40µs.
And this is due to the base functions not beeing on DMA - as they really do not need to be.
The huge boost I get is by sending the display-list-buffer as one chunk of data.
In case of the demo that is the 220 bytes shown in the display, these are sent in one go, CS low, address, 220 bytes, CS high.

>In the end the first reason was the more important reason as using LittlevGL for a complex >application was far, far easiest than using the co-processor for a complex GUI. 

As that may be, 4-5 FPS would not be useable for me.
I am always updating the display list 50x per second while trying to keep the time for the display-update under 1ms, that is reserving 5% of the runtime for EVE.
My last project needed a little more than that, around 1,2ms. But that was running on an ATSAMC21 which only is a Cortex M0 with 48MHz core clock and I was using 12MHz for the SPI.
Just to check it out I put it on the ATSAME51 now and time1 is 180µs for composing a CMD-FIFO of 1184 bytes.
With a time budget of 20ms this is using less than 1% of it.
This is just what EVE is designed for, off-loading the HMI by composing it from objects instead of pushing pixels.
And of course this is with touch under EVEs control.

>Plus the code is smaller in size overall now, both as written and compiled.

That is not so much a concern for me as the runtime is.
The project I am looking at now builds to 43540 bytes.
That includes a couple of images and a whole lot of strings.
When I remove most of the error condition strings it already drops to 24924 bytes.
This still contains more strings and about 5k for the images.
So data really is what makes the binary size grow.
But the ATSAMC21 I use has 256k FLASH and the ATSAME51 has 512k FLASH.
The ESP32 starts at 4MB.

And then there are the BT815/BT816 EVE chips, the display I am working with is an EVE3-50G.
It has a 16MB FLASH on board attached thru a dedicated QSPI interface that you can use to directly display images from and that is written to with a USB/SPI adapter.
The project does not make use of this as it ran on a FT813 based display we already had.

But one other project I did was using an Arduino mini pro - as requested by the client.
And it compiles to only 16k.
It uses several images on the 5" 800x480 display though, including an animation.
And it uses a size 140 UTF-8 font.
The binary for the display-FLASH is 6.2MB in size.

>Except, on the ESP32 I still need to let the driver manage the CS line for an entire transaction

Why?
Because the SPI driver you are using is doing it this way?

>And then you have to have the proper code anyway to wait for queue completion
>too even for single transfers the second you use DMA.
>This is because they use FreeRTOS queues internally and you really have
>to keep that internal queue in sync.

Okay, I am really not using a que, that is a huge difference.
But this que probably is software driven, using a list of buffers and an interrupt.

>I have no way to "prepare" a buffer for it to use and I really don't want to duplicate
>buffers just to prepend an address. 

Then how do you prepend the address?

>So I have to sent addresses for each transfer but I don't have to duplicate any buffers
>on the ESP32 because it can handle the separation of address
>and data in its SPI transfer API

That sounds fishy, as if the SPI function is manually setting CS, transferring the address without DMA and then the buffer by DMA.

> Furthermore, LvGL may decide to update a smaller rectangle of the screen.
>In those cases, each line of the bitmap has to be written to a different region of the EVE memory.

This sounds like a huge bottleneck.
Short DMA transfers do not make much sense, at some point it will take more time to setup the DMA transfer than it would take to just send a couple of bytes.
Sure, I can not tell in this case where the cutoff-size is, if it is at x bytes or at xx bytes.
With 240MHz core clock it should be on the lower side but that que most likely does come with a price as well.

>One issue is that the FT81x uses the status of the CS line to determine
>the end of memory mapped transfers

That is how SPI chips generally work, this is not a flaw in EVEs design.
And there are not many ways to handle this differently.
For example a timeout for resetting the state machine would really annoy me.
Image that you would need to wait for x µs between two different transfers.
And if that timeout is rather short this could provide a whole lot of fun with consecutive transfers beeing a little too slow because the host-controller had to serve an interrupt.

>On the ESP32, this means letting it manage the CS line for all transfers or you get into
>a weird state of trying to co-manage and how can you tell when it is safe to use it outside of DMA. >It quickly gets more complicated than what I have done.

I am suspicious that this more an attribute of a weird SPI library than of the ESP32 itself.

davidjade, 2020-05-16:
I agreed that a lot of what I did was likely off scope for EVE users, but if for instance you wanted to stream MJPEG to EVE for video display (using the so-processor) on an ESP32 using DMA, then maybe not. Hard to say for sure until someone else uses your DMA method on the ESP32 to compare with mine.

I am genuinely curious why you think the numbers are too slow for the ESP32. Set aside the simple EVE demo project for a second. If the screen is 800x480 -16-bit that is 6,144,000 bits of data. SPI is a serial protocol and it transmits one bit per clock (MOSI/MISO mode). That means at a 30Mhz SPI bus rate you can update the full screen exactly 4 times per second if you do full screen updates (30Mhz == 30,000,000 bits == 6,144,000 x 4). This is my understanding of SPI and the relationship to bus speed - maybe I am wrong. This ignores the little overhead in address, commands, dummy bytes, etc...  

This is the SPI throughput I get with my code. This is much more than the old code was capable of on the ESP32. On the ESP32 it required me to use the full set of features that the SPI DMA driver provides. Trying to manage the CS line, my own buffers that combine address bits + data bits, etc.. - this all slowed the ESP32 SPI transfer rate down significantly - I know because I tried that first - so I had to find a better way. I believe I have squeezed everything out the the ESP32 SPI bus driver that is possible because I am running at the full rate of the FT81x SPI bus where it really matters most - transferring large buffers of data at the maximum SPI speed as spec'd for the FT81x.

Maybe for small transfers, like in the demo, it is not the case that I am able to saturate the SPI bus but if so, then it is a problem with the ESP32 SPI DMA driver, outside anyone's (easy) control. Or maybe enabling queuing of multiple DMA transaction would help. Either way, more buffer copying/manipulation, single byte transfers, etc... they are all going to be slower as there is a cost in SPI cycles on the ESP32. In the end I can only run as fast as the ESP32 SPI driver allows but it is capable of running at 80Mhz if the slave supports that. My point is, on the ESP32 large gains can be made by reducing the number of transfer cycles and letting it manage the details, so that is what I did. That's really all my point is. 

Now you can say this is still too slow but that's a limitation of this method of updating the display since EVE sets the maximum SPI bus speed. I realize that. Again my only point was from a pure SPI bus perspective, I squeezed pretty near the maximum FT81x SPI bus performance when using an ESP32 with low overhead, both in memory use and CPU cycles. That I am using it to update the display in a bottle-necked way isn't really the point when talking about effective SPI bus transfer speed.

re: CS line management, addresses + data, etc...

The SPI library I am using is the one provided by Espressif, the makers of the ESP32. It is part of the ESP-IDF SDK and called "spi_master". Sure there is likely a lower-layer ESP32 SPI hardware interface but this is the driver they designed to drive DMA transfers at the full speed of the ESP32 SPI bus (80Mhz). I think it is quite nice to understand and use.

The spi_master driver in the ESP32 is really engineered to control the CS line itself - that's a decision that Esspressif Systems made, that's all I'm saying. While you can go it alone using manual CS control you can't really if you want to use DMA unless you want to use pre/post callbacks which just slow things down (or block waiting for DMA to complete to toggle CS - what is the point of that?). It is also clearly more performant if you just let it manage this and it also helps with keeping the address and data transmission in the same transaction without having to buffer them together because the SPI API allows you to specific an address, command, and data bytes all in one SPI transaction call.

You bundle address + command + data by filling out the structure that you pass for the SPI driver to do a SPI transfer (flags control the number of bits so you can use variable-bit addresses/commands, etc...)
``` c
struct spi_transaction_t {
    uint32_t flags;              
    uint16_t cmd;                // command bits
    uint64_t addr;               // address bits
    size_t length;                 // data length
    size_t rxlength;             
    void *user;                  
    union {                         // tx data
        const void *tx_buffer;   
        uint8_t tx_data[4];      
    };
    union {                         // rx data
        void *rx_buffer;         
        uint8_t rx_data[4];      
    };
} ;

typedef struct {
    struct spi_transaction_t base; 
    uint8_t command_bits;          
    uint8_t address_bits;          
    uint8_t dummy_bits;            
} spi_transaction_ext_t ;

```
So if you want to use the ESP32 SPI native driver you have to use these structures. They do ultimately just just send all the bits mashed together but they don't require me to mash them all together, which helps greatly in not having to do extra buffer manipulation - that's all I am pointing out. From this perspective I think this is a well-designed and flexible SPI driver.

This feature allows be to take a block of pure data and also specify the address and command to use without having to combine them first into one contiguous transfer buffer. That's all I am pointing out. Less buffer manipulation when you don' [... cut ...] of using EVE features too. I don't think the existing transparent DMA code would have allowed that _on the ESP32_ without causing double buffering and/or buffer manipulation or other changes to separate address + data bits as the ESP32 driver allows.

Anyway, it was only ever my intention to share what I did and highlight how to best use the native ESP32 DMA SPI driver, what I learned, and what I felt was necessary to get the most out of it. Thanks for your original code which helped me greatly.

RudolphRiedel, 2020-05-17:
> I am genuinely curious why you think the numbers are too slow for the ESP32.
> Set aside the simple EVE demo project for a second.
> If the screen is 800x480 -16-bit that is 6,144,000 bits of data.
> SPI is a serial protocol and it transmits one bit per clock (MOSI/MISO mode).

You are correct, but that is not what I meant.
You listed two different speeds, the 4-5 FPS for LittlevGL which I would consider unuseable.
But yes, I am aware that this is only the beginning and using single-line SPI.
I can not compare the LittlevGL framerate but you are correct, transferring raw data for 800x480x16 takes about 0.2 seconds at 30MHz single-line SPI.

LittlevGL is interesting indeed, but I would use it from a bigger controller than I am using now.
Maybe with a parallel memory interface into a framebuffer or with a gfx-controller.
I also can not really comment on that it is easier to create GUIs with LittlevGL, not after only a quick look at the demo source-codes on Github.
It sure is different and you most likely have more options for the GUI elements.

But for the small controllers I use EVE is an excellent companion.

And you listed the times from my demo code.
I can compare the demo times to what I see with my boards and I did crunch the numbers on that.
This would be highly useable, no doubt, but I would expect less than 30µs and not 190µs.
And 1586us is surprising when an Arduino Uno with 16MHz core and 8MHz SPI only needs 704µs.

I merely had a glimpse at the ESP32 so far, I designed a few boards for ESP32 modules but I did not program the ESP32 so far.
So I stick to what I know and that is what my code puts out on the SPI and how.

I guess I could give it a try myself but then I really do not have any use for the ESP32.

>My point is, on the ESP32 large gains can be made by reducing the number of transfer cycles
>and letting it manage the details, so that is what I did.

Well, the number of transfer cycles for my demo is 2 for time1 and DMA active.
EVE_memRead16(REG_CMD_DL); - to calculate the resulting display-list size, no DMA used
Secondly the 220 bytes of co-processor commands as one chunk of data with DMA.

That is followed by a call of EVE_cmd_start() when the DMA transfer is done.

So without the debug-information the function TFT_display() only does a single transfer for
all the commands that are send to the co-processor when DMA is used.
And it does not wait for anything to be finished.
The resulting time1 is then only the time it takes to fill the buffer.
The speed of the SPI does not really matter for this.

Have you checked the SPI with a logic analyzer?

>The SPI library I am using is the one provided by Espressif, the makers of the ESP32. 
...
>From this perspective I think this is a well-designed and flexible SPI driver.

And I would not argue that it is not well-designed or not flexible.
I am only saying that the numbers you provided for the EVE demo project are
lower than what I would expect.

I can't even tell if the ESP32 could do better or not.
And even then it most likely would not even matter since the 191µs is not a bad result.

>I don't think the existing transparent DMA code would have allowed that on the ESP32 without
>causing double buffering and/or buffer manipulation or other changes to separate
>address + data bits as the ESP32 driver allows. 

That is now purely outside of LvGL and does not deal with the issue that I am using DMA and non-DMA transfers.
But, just looking at the DMA portion, there is not much that it does.

EVE_start_cmd_burst() writes the address to the first three bytes of a buffer in case DMA is enabled.
That can easily be changed to write that address to some var and then EVE_dma_buffer_index could be set to Zero instead of 3.

#if defined (EVE_DMA)
 #if defined (ESP32)
   EVE_spi_adr = ftAddress; // may need some byte-order switching
   EVE_dma_buffer_index = 0;
  #else 
   EVE_dma_buffer[0] = (uint8_t)(ftAddress >> 16) | MEM_WRITE;
   ...
  #endif

If cmd-burst is active the commands no longer control CS and no longer add the address
to the transfers and they call spi_transmit_async().
In case DMA is enabled spi_transmit_async() only is:
EVE_dma_buffer[EVE_dma_buffer_index++] = data;

EVE_end_cmd_burst() just calls EVE_start_dma_transfer().
And what EVE_start_dma_transfer() does is up to you.

At the end of TFT_display() is EVE_cmd_start() and this is the only part where things get tricky.
I am using the var EVE_dma_busy to signal that a DMA transfer is in progress and so EVE_cmd_start() actually does nothing at all at this point.

However, the command co-processor still needs to be told that there is something in the FIFO to be executed by writing the new end-address to REF_CMD_WRITE.
I am using the DMA interrupt for this.
Set EVE_dma_busy to Zero, clear the chip-select line and call EVE_cmd_start().

I don't know, maybe the ESP32 SPI driver allows for a callback function that is called when a buffer is transferred.
That could set EVE_dma_busy to Zero and call EVE_cmd_start().

Well, that is not really something that you need, granted.
But up to and including transferring the command-FIFO buffer this does not interfere with your requirements, including control of the CS line.

davidjade, 2020-05-17:
Thanks for the insightful response.

For the time1/time2 on ESP32 Arduino, I have read through the ESP32 spi_master driver and there's not really a lot going on. However on their Arduino framework there is - a lot of dealing with setting up and tearing down the SPI bus config to enabling changing bus parameters on the fly. I think they try to be smart there but perhaps it is just the overhead of trying to be Arduino compatible. When you take the ESP32 Arduino framework out of the picture things improve dramatically, but by then you're also likely using DMA via the lower level SPI APIs too then, which has other benefits.

Still though, for whatever reason not batching all SPI calls as efficiently as possible on an ESP32 is not a great experience no matter it its Arduino or not. Sure all the setup up calls shouldn't matter as they only get called once but the effect is, the display does not wake up nearly as fast. I didn't measure this though as it was just so obvious. This is why I just decided to buffer everything but then I had to make allowances for not double-buffering larger transfers, sending addresses, etc...

I have looked at a logic analyzer and there are just little delays between separate SPI calls, again on Arduino the efficiency is really terrible. Also, I cannot explain why the Arduino version is only stable at 16Mhz while the full DMA version is stable at 30Mhz. Again there is some bad timing somewhere in the Arduino library provided by Espressif. It is great and all to have the Arduino compatibility for ESP32 but perhaps they don't (or can't) take all the necessary steps to make it as high performance as native - I don't really know.

That way I handle the FIFO and REF_CMD_WRITE is that my ESP32 DMA code has the ability to wait for all DMA transfers to complete. It does this by deferring to the driver, which waits on the transfer queue to be empty. Since it is using a FreeRTOS Queue internally, this wait should be very efficient since it puts the thread into a wait state on a waitable object - i.e. more efficient than spinning on a boo as the CPU is then free to do other things. I leverage this in several places btw, i.e. during EVE setup code where you want to wait for something to be processed before continuing, etc... So in my version EVE_target needs to have a function defined that waits until transfers have completed. In the Arduino case it's a no-op since there is no DMA. For ESP32 it is deferred to the SPI driver. You could still use a flag that is set via a DMA interrupt too if other platforms don't have a native way to detect DMA completion. Or perhaps a Mutex/Semaphore waitable object if the platform provided that.

So perhaps I could have solved this by using your version of bursting and also changing all the setup code to use it too to solve the start up delays on ESP32 but at that point it seemed just as good to make it buffer/burst all SPI transfers (except very large) so that's what I did. In my version all SPI transfers are "bursted", just in a different, implicit way. It accomplishes the same thing in the end as your code, but now it happens everywhere and it's up the the final EVE_target code to decide what to do with the buffered/bursted SPI data - send it in an Arduino compatible way or use a native DMA API. It's just a buffer of data to send at that point, automatically batched as much as possible.

So I think our versions accomplish the same things, just in different ways. My version buffers/bursts all SPI transfers implicitly to help with the latency of ESP32 transfer cycles whereas your version only buffers/bursts when you explicitly call for it. Yours does lead to duplication of code since you have many places to detect whether to make async calls or normal calls throughout the EVE code whereas mine is transparent as the implicit buffering has been pushed down a level.

I really wish I had a few more boards like Uno and the others ones you are using for DMA. I would like to try my code with them. Not that I expect it to be really faster for those cases, just to ensure that it also worked in those cases. I can't see any reason why it would not, with some small changes, but I can't test this without more hardware.

closed 2021-01-18 by RudolphRiedel