RudolphRiedel/FT800-FT813 #11: Support for Newhaven Displays EVE2 part # NHD-4.3-800480FT-CSXP-CTP

melted99, 2020-07-08:
Does this library also support the Newhaven Displays part # NHD-4.3-800480FT-CSXP-CTP?

Best Regards,

Ted

RudolphRiedel, 2020-07-08:
Not directly under this name as a 4.3" with 800x480 is rather unique and I did not know these exist so far.
But I just checked the user-manual and the "EVE_NHD_70" parameter set should work.

daniel-streetec, 2020-07-09:
Hello,
I have tried using the NHD_70 Parameters and also tried creating my own parameters... The LCD part is working, so the display is no issue. The issue I have at the moment, is that touch is not working at all. 
Even reading the Raw Registers is return default values only.

Do you have an idea what I could try?

Thanks a lot

RudolphRiedel, 2020-07-09:
Have you tried calibration?

According to the datasheet the display uses a FT5426.
And according to AN_336 from Bridgetek this should just work with no additional code required.

daniel-streetec, 2020-07-10:
Hi, thanks for the answer.
Yes I tried the calibration. When I get to the calibration screen I see the dot flashing and I had it two times that the calibration was „working“ (tapping over and over again and eventually it recognized the touch) and after two times nothing happened again.
The calibration values differed a lot the two times the calibration „worked“.
According to the manufacturer a calibration is not needed for the capacitive version... but never the less it should work.
Am I Right, that the calibration is running without the mcu? So the conclusion would be, that there is either an issue with the configuration of the touch functionality or with the touch hardware(?)
I have two (new) displays of this type and really would like to implement them in a project of mine, but neither work

RudolphRiedel, 2020-07-10:
This really should not be an issue with cap-touch but try raising EVE_TOUCH_RZTHRESH to 1800.
Have you tried displaying REG_TOUCH_RZ?

And yes, you always have to calibrate your touch-controller.
While you can use the REG_TOUCH_TRANSFORM_x values across modules with the same panel to some extent,
you always need a set of values.
I am using a special stylus to get better calibration results.

The calibration is running without the mcu, correct.

daniel-streetec, 2020-07-13:
Hey,
I just tried reading REG_TOUCH_RZ and it also stays at the default value of 0x7FFF.

i would say, that there is something wrong with new NewHaven Hardware... Can you recommend a "high-resolution" touch panel with 4-5" and touch functionality - that is working...? Thanks a lot

RudolphRiedel, 2020-07-13:
It is weird that both displays are not working, looks more like systematic than defective.
Can you locate R19 and R20 on the display and check their values?
The last time I had a serious issue with touch the (different) manufacturer messed up these pullup-resistors.
Well, at least they are correct in the Newhaven schematic at 1k each.

My personal preference currently are the EVE3 xxG modules from Matrix Orbital,
a fitting one would be this: https://www.matrixorbital.com/ftdi-eve/eve-bt815-bt816/eve3-50g

Alternatively I would recommend getting on of these:
https://riverdi.com/product/ritft50cap/
https://riverdi.com/product/ritft50capfr/
https://riverdi.com/product/ritft50capux/

These do however use a completely different connector.
Well, Riverdi used it first, Matrix Orbital adapted it for their modules and Newhaven had for some reason make
theirs different by 1 pin and use the more chunky 1mm FFC.

Crystalfontz also has a 5" with 800x480:
https://www.crystalfontz.com/product/cfaf800480e1050sca11-800x480-eve-accelerated-tft
But it "only" uses a FT813 and I would prefer BT815 by now.

RudolphRiedel, 2020-07-13:
I checked for Newhaven repositories and found this:
https://github.com/NewhavenDisplay/EVE2-TFT-Modules/tree/master/4.3in/_800x480

These are derived from very old FTDI examples.
But just going over Ft_App_Gradient.ino I could not find anything special in regards of the touch.
And the parameters really are the same as NHD_70.

So if this works, so should my library.

daniel-streetec, 2020-07-15:
Hey,
thanks for the Info.

to complete this topic: I got everything working now with the NHD display as well. It seems there is some funky power supply stuff going on which is causing the touch issues.

If I connect the MCU, the boards logic and the backlight to the same 3V3 source, touch is not working, same is true if I only connect touch and logic to a separate 3V3 source and power the MCU from USB (gnd of course connected). If I power them from separate sources (MCU + logic from one source, backlight from the other) touch is working.
It seems that, as soon as I power the logic and the backlight from the same source, touch is not working.

RudolphRiedel, 2020-07-15:
Now that you mentioned it, I had something similar with a RVT70 from Riverdi but it was not the display but my power supply.
In this case it initially worked though but the touch controller eventually glitched out.

Looking at the manual for the NHD-4.3-800480 I am not sure what the power requirements for this one are.
The typical supply current of 40mA for the logic must be a mistake by Newhaven, more likely would be 80mA and 140mA max.
And BLVDD with 3.3/540mA or 5V/240mA looks like the backlight regulator is loosing efficiency at lower voltages.

dstulken, 2020-10-14:
If it helps, I can provide another data point.  
I was able to successfully run your library test app on the EVE2 (FT813) NHD-4.3-800x480FT-CSXP-CTP, including touch.

My setup:
- ESP32, running the Arduino code branch, but from the Arduino IDE instead of PlatformIO.  
_(It complained about "unrecognized characters", but the "Fix Encoding & Reload" option solved it with no other project changes.)_
- Using the EVE_NHD_70 profile
- Powering the microcontroller, the touchscreen logic, and the backlight each on separate 3.3v supplies.  I didn't do this because of the findings above, only because that's how my PCB power was already segmented for other power loading reasons.
- I do not have the PDN pin connected - Not sure if that matters for the result.   (This is an aside, but I left this pin disconnected as I will have no need to power down or reset the display once running.  If in your experience, EVE devices in general should be manually reset once at startup for proper operation, I'd be interested to know this!)

So yes, in short, the EVE_NHD_70 profile, along with separate power supplies, seems to work fine.    Perhaps changing the "EVE_NHD_43" profile definition to "EVE_NHD_43_480X272" and "EVE_NHD_43_800X480" (or something to that effect) in EVE_config.h to add the new option and disambiguate the old, would be a good solution?

I'd be happy to run any other tests if they would be helpful - just let me know.
Thanks!

RudolphRiedel, 2020-10-16:
Somehow the EVE_NHD_43_800480 profile is not in the V4 archive, something went wrong there.
But it is in the V5 archive: https://github.com/RudolphRiedel/FT800-FT813/blob/5.x/EVE_config.h
The history even has this as the last entry for V4.

I will switch to V5 as the main branch when the BT817/BT818 get released by Bridgetek in November.
Well, after I put everything back to support these first. :-)

closed 2020-10-30 by RudolphRiedel