RudolphRiedel/FT800-FT813 #12: Raspberry Pi support

metanav, 2020-07-09:
Does it work with Raspberry Pi?

RudolphRiedel, 2020-07-09:
No, so far not, you are the first to mention Raspberry Pi.

While it should be possible for someone with Raspberry Pi experience to add support for it,
I do not have any real experience with it or with programming under Linux.

Also I do believe that Raspberry Pi or Beaglebone and similar boards are out of scope for EVE.
The Raspberry Pi already allows connecting to a display directly and has a graphics controller built in, you can even use HDMI.

That said, check out BRT_AN025 from Bridgetek, this is a beta release for a new "Portable MCU library for EVE", the "Examples" folder has a "RaspberryPi" folder among others.

metanav, 2020-07-10:
Thanks @RudolphRiedel for pointing to the right direction. I was able to run basic example (in the source package) with Gameduino 3 shield using RPi 4.

closed 2020-07-10 by metanav

RudolphRiedel, 2020-07-10:
How did you manage to use the Gameduino 3 shield?
As far as I know there are no timing parameters provided for the Gameduino shields.

metanav, 2020-07-10:
It took a while to figure out correct parameters for Gameduino 3 (FT810 EVE).
```
#define EVE_DISP_WIDTH 480 // Active width of LCD display
#define EVE_DISP_HEIGHT 272 // Active height of LCD display
#define EVE_DISP_HCYCLE 548 // Total number of clocks per line
#define EVE_DISP_HOFFSET 43 // Start of active line
#define EVE_DISP_HSYNC0 0 // Start of horizontal sync pulse
#define EVE_DISP_HSYNC1 41 // End of horizontal sync pulse
#define EVE_DISP_VCYCLE 292 // Total number of lines per screen
#define EVE_DISP_VOFFSET 12 // Start of active screen
#define EVE_DISP_VSYNC0 0 // Start of vertical sync pulse
#define EVE_DISP_VSYNC1 10 // End of vertical sync pulse
#define EVE_DISP_PCLK 5 // Pixel Clock
#define EVE_DISP_SWIZZLE 3 // Define RGB output pins
#define EVE_DISP_PCLKPOL 1 // Define active edge of PCLK
```

RudolphRiedel, 2020-07-10:
Thank you, I'll add an profile for it.