RudolphRiedel/FT800-FT813 #20: Support for new EVE4x-70G IPS displays from Matrix Orbital

dstulken, 2021-05-25:
Hello!   I'm testing out one of the pre-production 7.0" IPS 1024x600 EVE4 displays from Matrix Orbital, but am having a bit of trouble getting it to function correctly with this library.  

The symptoms are that an image displays, but the whole screen has an intermittent flicker, sometimes with a position offset, as well as some vertical streaking primarily on the left edge (which appears as horizontal streaking on the top edge in my application, as I'm using rotation 2 for portrait orientation).  Touch appears to work OK for buttons, but none of the sliders are functional.  After a period of time (a minute or so), the display will go dark until power cycled.     If the code execution is interrupted, such as when the microprocessor is being re-programmed, the display (in its frozen state) is visually 100% stable, so it appears as though the glitching only happens while the display is actively being updated.

Being brand new, there wasn't a definition for this display in EVE_config.h, so I requested the timing parameters from Matrix Orbital and created my own.  I've never done this before, so I wouldn't be surprised if I made a mistake at this step, or misinterpreted the mapping of the timings for how this library expects them.

The timing values I received from Matrix Orbital:
``` 
DWIDTH = 1024;
DHEIGHT = 600;
PIXVOFFSET = 0;
PIXHOFFSET = 0;
HCYCLE = 1259;
HOFFSET = 85;
HSYNC0 = 37;
HSYNC1 = 40;
VCYCLE = 768;
VOFFSET = 16; 
VSYNC0 = 13;
VSYNC1 = 20;
PCLK = 1;
SWIZZLE = 0;
PCLK_POL = 0;
HSIZE = 1024;
VSIZE = 600;
CSPREAD = 0;
DITHER = 1;
```

And then my resulting configuration:
```
#if defined (EVE_EVE4_70G_TEST)
#define EVE_HSIZE  (1024L)
#define EVE_VSIZE (600L)

#define EVE_VSYNC0  (13L) 
#define EVE_VSYNC1  (20L)
#define EVE_VOFFSET (16L)
#define EVE_VCYCLE  (768L)
#define EVE_HSYNC0  (37L) 
#define EVE_HSYNC1  (40L) 
#define EVE_HOFFSET (85L)
#define EVE_HCYCLE  (1259L) 
#define EVE_PCLKPOL (0L)
#define EVE_SWIZZLE (0L)
#define EVE_PCLK  (1L)  
#define EVE_CSPREAD (0L)
#define EVE_TOUCH_RZTHRESH (1200L)

//#define EVE_HAS_GT911  /* special treatment required for out-of-spec touch-controller */
#define EVE_HAS_CRYSTAL
#define FT81X_ENABLE
#define BT81X_ENABLE
#endif
```

As you can probably tell from those defines, I'm still on the 4.x branch from the end of last year.  If supporting this display requires an upgrade to the 5.x branch I can do that, I'm just trying to not change too many things at once if I can avoid it.

Thanks in advance for the help!

RudolphRiedel, 2021-05-25:
Unfortunately Matrix Orbital has not released any documents so far with parameters and the EVE4x displays are not sold by distributors so I can not buy one of these yet.
The timing parameters for the EVE4-101 and EVE4-40 are from their software.

That said, I find the parameters you received for the EVE4x-70G a bit odd.
Which does not mean that these are wrong but the other 1024x600 panels I have in my spreadsheet have other parameters.
One value that is particulary odd in this set is PCLK = 1 though.
For the BT817/BT818 this has a special meaning, this enables the second PLL - EXTSYNC.
Without setting this up and the 4.x branch definately does not do that, REG_PCLK_FREQ has
the default value of 60MHz which should result in 62 frames per second.
This may explain the other timing values, maybe they went back from this default value of 60MHz and adapted
the values to get as close as possible to 60 FPS.
Instead of using the timing values for the panel and adjusting the frequency.

A configuration based on the timing values given for the 5.x branch could be this:

#if defined (EVE_EVE4_70G)
#define EVE_HSIZE  (1024L)
#define EVE_VSIZE  (600L)
#define EVE_VSYNC0  (13L)
#define EVE_VSYNC1  (20L)
#define EVE_VOFFSET  (16L)
#define EVE_VCYCLE  (768L)
#define EVE_HSYNC0  (37L)
#define EVE_HSYNC1  (40L)
#define EVE_HOFFSET  (85L)
#define EVE_HCYCLE   (1259L)
#define EVE_PCLK  (1L)  /* 1 = use second PLL for pixel-clock in BT817 / BT818 */
#define EVE_PCLK_FREQ (59000000L)	/*  59MHz - value for EVE_cmd_pclkfreq -> 61 FPS */
#define EVE_PCLKPOL (0L)
#define EVE_SWIZZLE (0L)
#define EVE_CSPREAD  (0L)
#define EVE_TOUCH_RZTHRESH  (1200L)
#define EVE_HAS_CRYSTAL
#define EVE_GEN 4
#define EVE_HAS_GT911
#endif

So pretty much the same.
Regarding the GT911 I can only guess as I do not have data if the EVE4 actually are still using these.

I believe I have seen the panel without the BT81x board at the Matrix Orbital website, if I find it I'll have
a look at the timing parameters.

You could of course use the 4.x branch but it does not support the newer commands and registers, as few as these are.
And it is slower.

dstulken, 2021-05-25:
Thanks for the info and the quick reply.

I did a test with the 5.x branch.  It took me a few minutes to unravel the new Arduino/ESP32 changes and figure out what I needed to change in EVE_target.h, but after doing so, the display actually came up just fine using the 5.x configuration above.  No flicker, no streaks, no offsets, and touch works (tapping the "Touch!" button starts/stops the star animation).   In my prior 4.x-based application, I had noticed that button touches worked but sliders did not - I'm not sure if that is still the case - are there any trackers set up in the test page?

I tried commenting out EVE_HAS_GT911, and after power cycling, the touch no longer works.  (It *did* work after the software change and ESP32 reset, but prior to power-cycling the display, so that must be setting some register that was persisting.)  But yes, I can confirm that EVE_HAS_GT911 is still required.   (Is the GT911 part of the touch panel?  It would make sense if this new display used the same adhesive wide bezel touch panel as the prior units...)

Soo.... positive result, but still interesting.  Is it likely to be that missing EVE_PCLK_FREQ functionality causing my issues in the 4.x branch?

Thanks again for your help!

dstulken, 2021-05-26:
> I believe I have seen the panel without the BT81x board at the Matrix Orbital website, if I find it I'll have
> a look at the timing parameters.

Looks like it might be this one?
https://www.matrixorbital.com/mop-tft1024600-70g-ips

RudolphRrr, 2021-05-26:
Nice, that should be it and I could not find it yesterday way past my bedtime. :-)
And the timing values are much more like I would expect them to be with HCYLCE = 1344 and VCYCLE = 635.
But well, the given values are within the limits of the datasheet and it works now. :-)

>I had noticed that button touches worked but sliders did not - I'm not sure if that is still the case
>- are there any trackers set up in the test page?

Yes, sliders need updating thru tracking and this is not part of my basic example.

RudolphRiedel, 2021-05-26:
Please try this one:

#if defined (EVE_EVE4_70G)
#define EVE_HSIZE (1024L)
#define EVE_VSIZE (600L)
#define EVE_VSYNC0 (0L)
#define EVE_VSYNC1 (10L)
#define EVE_VOFFSET (23L)
#define EVE_VCYCLE (632L)
#define EVE_HSYNC0 (0L)
#define EVE_HSYNC1 (70L)
#define EVE_HOFFSET (160L)
#define EVE_HCYCLE (1344L)
#define EVE_PCLK (1L) /* 1 = use second PLL for pixel-clock in BT817 / BT818 /
#define EVE_PCLK_FREQ (51000000L) /* 51MHz - value for EVE_cmd_pclkfreq -> 60 FPS */
#define EVE_PCLKPOL (0L)
#define EVE_SWIZZLE (0L)
#define EVE_CSPREAD (0L)
#define EVE_TOUCH_RZTHRESH (1200L)
#define EVE_HAS_CRYSTAL
#define EVE_GEN 4
#define EVE_HAS_GT911
#endif

This is more based on the "Typ." values from the datasheet of the panel.
With a bit of guessing. :-)

RudolphRiedel, 2021-06-01:
Ping.
Did you have a chance to try out the updated profile?

dstulken, 2021-06-02:
Sorry, it's been a busy week!
Yes, I can confirm that that new profile with the "typical" timings works as well.

Something interesting though - if switching between the two profiles and reprogramming the microcontroller, the display will remain black after the microcontroller reset - it takes a power cycling (unplugging power and reconnecting it) for it to come up again, after which it displays correctly.  It happens in both directions too - switching from the first profile to your new one, as well as from new to old.   Resetting the microcontroller after this, everything is still fine - it is only when the profile changes that it doesn't work.

This isn't necessarily a problem - I don't ever anticipate changing display timings on the fly.    But does it point to something else? Perhaps a register that has an assumed state, that isn't being cleared or reset properly on startup when the library initializes?

One other interesting thing - I've been working on porting my "real" application into your 5.x codebase.  In the process, I discovered that the display wasn't actually displaying quite perfectly - it still had some horizontal streaks in places.  For whatever reason, the white background of your test app didn't show this, but my dark background does.   I did some experimenting, and changing EVE_PCLKPOL from 0L to 1L solved it.    My initial values came directly from Matrix Orbital via email - is there anything you see in the display data sheet that would indicate that this alternate polarity would have been expected?

dstulken, 2021-06-02:
Oh, that one is probably on me.  I have a simple RC time circuit to set that power-down/reset line, so that would NOT be resetting when the microcontroller resets.   It's still interesting that the display only fails when the timing parameters have changed though? Any other time, you can re-program or reset the microcontroller as many times as you want, and the display remains fine...

I'll try to get some photos when I have a chance.  In the meantime, based on the data sheet and my test results, do you think the "typical" value configuration would be the recommended profile to run for this display?  (and presumably, the version to add to the library for others to use?)

Thanks again for the help!

RudolphRiedel, 2021-06-02:
While the intended use is to use PD with a controlled I/O line, simply to bring EVE to a default state before configuring it, you can issue a software-reset as well.
There is this line near the beginning of EVE_init():
/*	EVE_cmdWrite(EVE_CORERST,0); */ /* reset, only required for warm-start if PowerDown line is not used */

Just activate it and the switching should work.

And regarding the profile, I just went ahead and added EVE_EVE4_70G yesterday, my second version with EVE_PCLK_FREQ,
I just feel this is closer to the intended use of the panel.
I also changed the EVE_PCLKPOL to 1 and am about the upload the next revision including a few other changes.

dstulken, 2021-06-03:
[image]

OK, here is the PCB.   Remember that this is a pre-production engineering sample.  I've been told that the production units will be a little bit different.

It's a pretty substantial departure from the tiny PCB on their prior 7" display, the EVE2-70G.  I think they decided to just make one PCB this time, that supports all of their different options, connections, and configurations in one go.

EVE2-70G for reference:
[image]

RudolphRrr, 2021-06-04:
>OK, here is the PCB. Remember that this is a pre-production engineering sample.

Thank you, so there is no LVDS transmitter.

>I've been told that the production units will be a little bit different.

Details I presume and I kind of like it, the layout makes a whole lot more sense to me than the layout of the 10" version.

The panel connector is off by maybe 5mm, that would be something to fix.
And I would shrink the PCB by a few mm to make it a bit smaller than the metal cage around the panel.
As it is the mounting area with the double-sided tape is reduced by the overlapping PCB.

It won't fit in the case I currently have for 7" panels but it should be easy enough to change the case.

>It's a pretty substantial departure from the tiny PCB on their prior 7" display, the EVE2-70G.

The reference would be the EVE3-70G in this case as it also has the external flash memory chip.
But the PCB is not much larger than the one the EVE2-70G has.

>I think they decided to just make one PCB this time, that supports all of their different options,
>connections, and configurations in one go.

One PCB is rather unlikely since the version without the cover-glas has the mounting-holes sticking out from the side.
But that would mean they still could use one layout on two different PCBs with the second one simply a bit larger.