RudolphRiedel/FT800-FT813 #38: Tag releases rather than changing default branch

j-c-cook, 2022-01-17:
I notice that the current branch is `5.x`, and that the `3.x` and `4.x` branches are behind those. I am assuming that each time a new "version" is being developed that you switch to a new branch and change the default. Why not work off of one default `master` branch and tag [releases](https://github.com/RudolphRiedel/FT800-FT813/releases) as you go? With git [tags](https://git-scm.com/book/en/v2/Git-Basics-Tagging), you can always checkout a previous version. (Creating a Github release will also create a tag.)

RudolphRiedel, 2022-01-17:
That is solely because my GIT skills are still poor but I am slowly working on it. :-)

j-c-cook, 2022-01-18:
Understood.

closed 2022-01-18 by j-c-cook