RudolphRiedel/FT800-FT813 #47: cmd_text does not display text

SirVer, 2022-07-15:
I am playing with a BT815 device and everything seems to be going great. Except for `cmd_font` which has peculiar issues. I tried V 4.0 of this library and also the newer that does not keep track of the `cmd_offset` anymore, instead writing into `REG_CMDB_WRITE`.

What I observe is that strings of certain length are not rendered, but the device continues. In my true application, later commands then sometimes report ominous errors like `invalid padding for cmd_flashread` - even though I do not intend to run cmd_flashread anyways. It feels like an alignment error, but I cannot pin it down.

Most lengths string work fine. If I run this python program as outside trigger:

```
for i in range(1, 100):
    print i
    subprocess.check_call(["./tester", "a" * i])
    time.sleep(0.5)
```

And here is the full C program `tester`, without `#includes`:

```
int main(int argc, char** argv)
{
        if(platform_spi_init() != 1){
                printf("spi init failed");
                return -1;
        }
        if (EVE_init() == 0) {
                printf("eve init failed");
                return -1;
        }

        EVE_cmd_dl(CMD_SYNC);
        EVE_cmd_dl(CMD_DLSTART);

        EVE_cmd_dl(DL_COLOR_RGB | 0xffffff);

        EVE_cmd_dl(DL_CLEAR | CLR_COL | CLR_STN | CLR_TAG);
        EVE_cmd_dl(DL_CLEAR_RGB | 0x000000);

        EVE_cmd_text(100, 150, 23, 0, argv[1]);

        EVE_cmd_dl(DL_DISPLAY);
        EVE_cmd_dl(CMD_SWAP);
        EVE_cmd_execute();

        return 0;
}
```

All strings render, except for length 
```
12, 13, 14, 15
44, 45, 46, 47
76, 77, 78, 79, 
```.

However when I start with `i=44` instead of `i=1`, then the first string will render, but the next one will not. Can you reproduce this issue? Any idea what it might be?

RudolphRiedel, 2022-07-16:
This looks like the padding for the strings fails, strings need to be filled with zeroes at the end to full 32 bits.
But this really should not happen and the way it fails is really strange.
What target are you compiling for?

SirVer, 2022-07-16:
The host is an armv7l GNU/Linux, it is [specifically a Linux industrial computer for charging stations for electric vehicles](https://in-tech-smartcharging.com/products/charging-station-communication/charge-control-c), 1GB Ram and . It most closely resembles a raspberry pi.

~~~
$ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 12.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: Freescale i.MX6 UltraLite (Device Tree)
Revision	: 0000
Serial		: 52222
~~~

RudolphRiedel, 2022-07-16:
Ok, mit einer APU habe ich bisher noch gar nicht gespielt, das Ding hat ja schließlich auch eine eingebaute Grafik-Einheit und ist damit so ein wenig ausserhalb der Ziel-Gruppe. :-)
Aber eigentlich ist das auch "nur" ein ARMv7 und 32 bit, damit ist der dem Cortex M4 nicht ganz unähnlich.
Und Cortex M4 benutze ich ja hauptsächlich, damit kann das erstmal nicht was grundsätzlich inkompatibles sein.
Kannst Du einen Trace von den CMD_TEXT ziehen? Oder alles für ein Kommando irgendwie speichern?

SirVer, 2022-07-17:
@RudolphRiedel Danke für deine Unterstützung! Sehr gerne liefer ich alles, aber was genau meinst du mit "Trace"? Einen strace? Oder soll ich alles was über in `spi_transmit_with_len` über den SPI bus geschickt wird als hexcode dumpen? Kann ich heute abend liefern.

SirVer, 2022-07-17:
Hier ist ein `strace` [mit 11 buchstaben](https://gist.github.com/9642431a280d22a29e4e91ee32174802) (die angezeigt werden) und [einer mit 12 buchstaben](https://gist.github.com/73ff1372d125bf1dbaec8bb62953c33c) (die nicht angezeigt werden).

RudolphRiedel, 2022-07-17:
Moin, ich musste erstmal Google fragen was ein strace ist. :-) Linux und ich sind noch keine Freunde.
Und da war ich etwas langsam beim Tippen, dann schaue ich mir gleich mal die beiden strace an.

Was ich  zum Debuggen normalerweise mache ist mir den Datenstrom auf dem SPI mit einem Logic-Analyzer anzusehen.
Ich mache eine Aufzeichnung / Trace von so 2s ab Reset und schaue mir damit sozusagen das Endergebnis an, das muss zum einen für alle Platformen gleich sein, zum anderen kann ich daran das konkrete Timing Verhalten bewerten.
Also wenn die Funktion spi_transmit_with_len() mit einem CMD_TEXT gefüttert wird dann könnte der Hex-Dump davon schon interessant sein, wobei die ja nicht von mir ist, da ist die Frage woher der Puffer und die Länge kommen mir der die gefüttert wird und was die Funktion wirklich macht.

Dann habe ich gerade mal ein Diff gemacht. Benutzt Du den aktuellen Stand, oder den 5.0.4 Release?
Im aktuellen Stand habe ich die Funktion private_string_write() noch mal überarbeitet.
In der V4 ist die entscheidende Funktion EVE_write_string(), die sieht da noch mal anders aus.
Getestet sind eigentlich alle Varianten, aber ich glaube ich setze noch mal einen umfangreicheren Test dafür auf.

RudolphRiedel, 2022-07-17:
Also was hier drin steht wäre interessant:
ioctl(3, SPI_IOC_MESSAGE(32), 0x7edb3b00) = 32

Wobei, wenn 32 die Länge ist, dann ist das ja schon mal durch 4 teilbar.
Was mich aber wundert, das sollten 28 sein mit einem String von 12.
Und 24 mit einem String von 11.

Das Kommando selber braucht 12 Bytes und bei 11+ Null-Byte sind es eben 24, bei einer Länger von 12 kommen 4 Null-Bytes dazu.
Sind das einzelne Kommandos mit jeweills Chip-Select? Das würde die zusätzlichen Bytes erklären.
Dann müssten die Längen aber alle ungerade sein, weil die Adresse ja nur 3 Byte lang ist.

SirVer, 2022-07-17:
> Was ich zum Debuggen normalerweise mache ist mir den Datenstrom auf dem SPI mit einem Logic-Analyzer anzusehen.

Uh, das übersteigt meine Fähigkeiten. Auf der Arbeit haben wir aber so ein Ding, vll kann ein Kollege helfen. Das braucht dann aber sicher ein paar Wochen.

> Dann habe ich gerade mal ein Diff gemacht. Benutzt Du den aktuellen Stand, oder den 5.0.4 Release?
Das ist ein älterer stand, von Version 4. Ich hatte auch den 5.x mal am laufen, sah aber das gleiche verhalten.

> Sind das einzelne Kommandos mit jeweills Chip-Select? Das würde die zusätzlichen Bytes erklären.
Ich verstehe die Frage leider nicht - ich weiss nicht was ein Chip select ist.

So wie ich den code verstehe schreibt er 3 bytes für die adresse auf dem chip, dann das CMD_TEXT commando (4 bytes), 8 bytes für die argumente, dann den string mit nem nullbyte (12 + 1 bzw 11 + 1). Dann wird der content auf 4 bytes gepadded für den command buffer im device (der will ja 4 bytes alignment), dann wird der ganze buffer noch für SPI transfer gepadded (auf 4 bytes), ergo

~~~
string länge 11: `padded(3 (addr) + padded(4 (CMD_TEXT) + 8 (args) + 12 (str mit null))) = 28 bytes`
string länge 12: `padded(3 (addr) + padded(4 (CMD_TEXT) + 8 (args) + 13 (str mit null))) = 32 bytes` 
~~~

RudolphRiedel, 2022-07-17:
> So wie ich den code verstehe schreibt er 3 bytes für die adresse auf dem chip, dann das CMD_TEXT commando (4 bytes), 8 bytes für die argumente, dann den string mit nem nullbyte (12 + 1 bzw 11 + 1). Dann wird der content auf 4 bytes gepadded für den command buffer im device (der will ja 4 bytes alignment), dann wird der ganze buffer noch für SPI transfer gepadded (auf 4 bytes), ergo
> 
> ```
> string länge 11: `padded(3 (addr) + padded(4 (CMD_TEXT) + 8 (args) + 12 (str mit null))) = 28 bytes`
> string länge 12: `padded(3 (addr) + padded(4 (CMD_TEXT) + 8 (args) + 13 (str mit null))) = 32 bytes` 
> ```
> 
> Oder ist da was falsch?

Das Nullbyte für das Ende des Strings ist Teil des Paddings, es werden 1 bis 4 Nullbytes angehängt um auf volle 32 Bit für das Kommando zu kommen.
Ein einzelnes Kommando oder auch im Burst-Modes eine Folge von Kommandos muss leider immer eine ungerade Anzahl an Bytes haben.
Ich mache das so das ich von dem DMA Puffer das Nullte Byte nicht schicke, damit sind dann die ganzen Kommandos wieder auf 32 Bit aligned.

RudolphRiedel, 2022-07-17:
So sieht das im Logic-Analyzer aus, 11 Zeichen:
[image]

12 Zeichen:
[image]

Also 27 und 31 Bytes, oder eben 3+24 und 3+28.

SirVer, 2022-07-17:
Mmh.. wenn ich so die commands schicke, dann kommt gar nichts bei meinem device an. Das sehe ich daran, dass `cmd_write` nicht erhöht wird. Ich hab mein program noch mal so verändert, dass ich das padding über die kommandozeile anhängen kann. Also ich push `x y font opts text \x00` and dann noch mal so viele null-bytes, wie ich über die Kommandozeile bestellt habe. Dann habe ich experimentiert und aufgeschrieben, wie viele manuelle nullbytes ich pushen muss, dass text angezeigt wird. 

Ausserdem rufe ich cmd_font nun zweimal auf, weil mir aufgefallen ist, dass es Situationen gibt indem einmal aufrufen geht, aber das zweit mal im gleichen program nicht - da wird dann kein text angezeigt, aber cmd_write erhöht sich wie beim ersten Aufruf.

Dabei kam das raus:
```
stringlänge anzahl nullbytes
1  3
2  2
3  1
4  4   0 geht einmal, sieht komisch aus
5  3
6  2
7  1
8  4   0 geht einmal, sieht komisch aus
9  3
10 2
11 1
12     0 geth einmal (falsches zeichen), 8 geht einmal (ohne falsches zeichen), 12, 16, 20, 24 geht einmal (ohne falsches Zeichen)
13     7 geht einmal, 11 geht einmal, 15 geht einmal, jeweils ohne falsches Zeichen
14     6, 10, ... gehen einmal, jeweils ohne falsches Zeichen
15     5, 9, 13, ... gehen einmal, jeweils ohne falsches Zeichen
16 4 
17 3   
18 2
19 1
20 0

```
Mit komisch meine ich, dass nach dem letzten buchstabe noch kuddelmuddel angezeigt wird. Als ob das nullbyte des textes es nicht in den command buffer geschafft hat.

[image]

Hier sind einige hexdumps der geschriebenen bytes:

```
extra zeichen:
 8 pad 0: b0 25 78     0c ff ff ff     64 00 64 00     1c 00 00 00     30 31 32 33     34 35 36 37     00
12 pad 0: b0 25 78     0c ff ff ff     64 00 96 00     1c 00 00 00     30 31 32 33     34 35 36 37     38 39 61 62     00

funktioniert:
 8 pad 4: b0 25 78     0c ff ff ff     64 00 64 00     1c 00 00 00     30 31 32 33     34 35 36 37     00 00 00 00     00
16 pad 4: b0 25 78     0c ff ff ff     64 00 96 00     1c 00 00 00     30 31 32 33     34 35 36 37     38 39 61 62     65 66 67 68     00 00 00 00     00

# Geht nicht
12 pad 4: b0 25 78     0c ff ff ff     64 00 64 00     1c 00 00 00     30 31 32 33     34 35 36 37     38 39 61 62     00 00 00 00     00

# Geht einmal ohne falsches Zeichen, aber nicht zweimal
12 pad 8: b0 25 78     0c ff ff ff     64 00 96 00     1c 00 00 00     30 31 32 33     34 35 36 37     38 39 61 62     00 00 00 00     00 00 00 00 00
```

Hast Du noch irgendeine Idee?

RudolphRiedel, 2022-07-17:
Nun, die Hex-Dumps sind alle falsch, das kann ich hier so nicht beobachten.
Also selbst die wo es erstmal zu funktionieren scheint sind falsch.

8 Zeichen: b0 25 78 0c ff ff ff 64 00 64 00 1c 00 00 00 31 32 33 34 35 36 37 38 00 00 00 00
12 Zeichen:  b0 25 78 0c ff ff ff 64 00 64 00 1c 00 00 00 31 32 33 34 35 36 37 38 39 30 61 62 00 00 00 00
16 Zeichen:  b0 25 78 0c ff ff ff 64 00 64 00 1c 00 00 00 31 32 33 34 35 36 37 38 39 30 61 62 63 64 64 66 00 00 00 00

Die werden alle am Ende mit 4 Null-Bytes aufgefüllt und das ist auch richtig so.
Der Hintergrund ist das die Kommandos alle auf 4-Bytes aligned im Speicher anfangen müssen.

Bleibt die Frage, welcher Code ist das genau, poste doch mal bitte die EVE_cmd_text() function und die dazugehörige Hilfs-Funktion die Du konkret verwendest.

RudolphRrr, 2022-07-18:
Na, "fully contained" geht wahrscheinlich zu weit, mir war schon klar, dass Du meine Library nicht einfach nur benutzt, sondern da was eigenes daraus gemacht hast.
Da ist ja auch nichts gegen einzuwenden, das macht nur die Analyse praktisch unmöglich.
Bei allem was ich bisher getestet habe funktioniert mein EVE_cmd_text() so wie es soll - ich erhöhe gerade die Test-Tiefe.

Die Variante die Du gewählt hast, so mit Chip-Select und Adresse für jedes Kommando, ist übrigens die langsamste überhaupt.
Und das EVE_cmd_dl(CMD_SYNC); am Anfang bewirkt in dem Kontext gar nichts.

SirVer, 2022-07-18:
> Na, "fully contained" geht wahrscheinlich zu weit, mir war schon klar, dass Du meine Library nicht einfach nur benutzt, sondern da was eigenes daraus gemacht hast.

ne, eigentlich haben wir nur einen workaround für dieses alignment problem zwischen 12 und 16 zeichen gemacht. Ich glaube, sonst haben wir nicht viel verändert. 

> Bei allem was ich bisher getestet habe funktioniert mein EVE_cmd_text() so wie es soll - ich erhöhe gerade die Test-Tiefe.

Mein bauchgefühl sagt mir, dass es ein SPI kernel bug oder ein problem mit unserem spezifischen setup hier ist - ist ja schon speziell mit diesem charge controller als computer. Mein team meinte auch, dass mit dem logic analyzer könne wir hinkriegen.

> Die Variante die Du gewählt hast, so mit Chip-Select und Adresse für jedes Kommando, ist übrigens die langsamste überhaupt.

Geschwindigkeit ist mir im moment nicht so wichtig... Aber es gibt noch den burst modus, richtig? 

> Und das EVE_cmd_dl(CMD_SYNC); am Anfang bewirkt in dem Kontext gar nichts.

ja, ich glaube das habe ich jetzt auch verstanden. Das braucht man nur, wenn man zum beispiel den RAM mit anderen bildern updaten will und das nicht flickern soll, oder? Für display list commands sorgt ja `CMD_SWAP` schon dafür, dass die liste sauber ausgetauscht wird.

RudolphRrr, 2022-07-18:
So, ich habe mal ein EVE3-50G mit Pins für den Logic-Analyzer ausgestattet und ein vorhandenes Projekt mit dem aktuellen Stand aus Github gefüttert, sowie die Display-Liste angepasst.

Das hier wird ohne zu Murren gerade 50 Mal die Sekunde angezeigt:
[image]

`
EVE_cmd_dl(CMD_DLSTART); /* start the display list */
EVE_cmd_dl(DL_CLEAR_RGB | WHITE); /* set the default clear color to white */
EVE_cmd_dl(DL_CLEAR | CLR_COL | CLR_STN | CLR_TAG);
EVE_cmd_dl(TAG(0));
EVE_cmd_append(MEM_DL_STATIC, num_dl_static);
EVE_color_rgb(BLACK);
EVE_cmd_text(20, 100, 29, 0, "1234567");
EVE_cmd_text(20, 130, 29, 0, "12345678");
EVE_cmd_text(20, 160, 29, 0, "123456789");
EVE_cmd_text(20, 190, 29, 0, "1234567890");
EVE_cmd_text(20, 220, 29, 0, "1234567890a");
EVE_cmd_text(20, 250, 29, 0, "1234567890ab");
EVE_cmd_text(20, 280, 29, 0, "1234567890abc");
EVE_cmd_text(20, 310, 29, 0, "1234567890abcd");
EVE_cmd_text(20, 340, 29, 0, "1234567890abcde");
EVE_cmd_text(20, 370, 29, 0, "1234567890abcdef");
EVE_cmd_text(20, 400, 29, 0, "1234567890abcdefg");
EVE_cmd_number(60, EVE_VSIZE - 15, 26, EVE_OPT_RIGHTX|4, display_list_size); /* number of bytes written to the display-list by the command co-pro */
EVE_cmd_number(130, EVE_VSIZE - 15, 26, EVE_OPT_RIGHTX|4, cmd_fifo_size); /* number of bytes written to the cmd-fifo over the spi without adressing overhead */
EVE_cmd_number(220, EVE_VSIZE - 15, 26, EVE_OPT_RIGHTX|4, num_profile_a); /* duration in us of TFT_loop() for the touch-event part */
EVE_cmd_number(320, EVE_VSIZE - 15, 26, EVE_OPT_RIGHTX|4, num_profile_b); /* duration in us of TFT_loop() for the display-list part */
EVE_execute_cmd();
`

Und aus dem SPI kommt laut Logic-Analyzer gerade das hier:
name	type	start_time	duration	mosi	miso
SPI	enable	8.53623192e-08	1.99999999e-08		
SPI	result	8.25362319e-07	6.4e-07	0x30	0x00
SPI	result	2.12536232e-06	6.4e-07	0x25	0x4A
SPI	result	3.46536232e-06	6.4e-07	0x74	0x43
SPI	result	4.82536232e-06	6.4e-07	0x00	0x42
SPI	result	6.10536232e-06	6.4e-07	0x00	0xFC
SPI	result	7.44536232e-06	6.4e-07	0x00	0x0F
SPI	disable	8.70536232e-06	1.99999999e-08		
SPI	enable	1.03053623e-05	1.99999999e-08		
SPI	result	1.10453623e-05	6.6e-07	0x30	0x00
SPI	result	1.23453623e-05	6.4e-07	0x21	0x4A
SPI	result	1.37053623e-05	6.4e-07	0x00	0x43
SPI	result	1.50653623e-05	6.4e-07	0x00	0x42
SPI	result	1.63253623e-05	6.4e-07	0x00	0xB4
SPI	result	1.76853623e-05	6.4e-07	0x00	0x04
SPI	disable	1.89453623e-05	1.99999995e-08		
SPI	enable	2.00653623e-05	1.99999999e-08		
SPI	result	2.04453623e-05	6.4e-07	0xB0	0x00
SPI	result	2.17253623e-05	6.4e-07	0x25	0x4A
SPI	result	2.30853623e-05	6.4e-07	0x78	0x43
SPI	result	2.44853623e-05	6.4e-07	0x00	0x42
SPI	result	2.58253623e-05	6.4e-07	0xFF	0x00
SPI	result	2.72053623e-05	6.4e-07	0xFF	0x00
SPI	result	2.85053623e-05	6.6e-07	0xFF	0x00
SPI	disable	2.97853623e-05	1.99999999e-08		
SPI	enable	3.07853623e-05	1.99999999e-08		
SPI	result	3.11853623e-05	6.4e-07	0xB0	0x00
SPI	result	3.24453623e-05	6.4e-07	0x25	0x4A
SPI	result	3.38053623e-05	6.4e-07	0x78	0x43
SPI	result	3.52253623e-05	6.4e-07	0xFF	0x42
SPI	result	3.65653623e-05	6.4e-07	0xFF	0x00
SPI	result	3.79253623e-05	6.4e-07	0xFF	0x00
SPI	result	3.92453623e-05	6.4e-07	0x02	0x00
SPI	disable	4.05053623e-05	1.99999999e-08		
SPI	enable	4.15253623e-05	1.99999999e-08		
SPI	result	4.19053623e-05	6.4e-07	0xB0	0x00
SPI	result	4.31853623e-05	6.4e-07	0x25	0x4A
SPI	result	4.45453623e-05	6.4e-07	0x78	0x43
[... 656 lines cut ...]
SPI	disable	0.000930345362	2.00000004e-08		
SPI	enable	0.000932045362	1.99999999e-08		
SPI	result	0.000932785362	6.6e-07	0x30	0x00
SPI	result	0.000934085362	6.4e-07	0x25	0x4A
SPI	result	0.000935445362	6.4e-07	0x74	0x43
SPI	result	0.000936785362	6.6e-07	0x00	0x42
SPI	result	0.000938065362	6.4e-07	0x00	0xE4
SPI	result	0.000939425362	6.4e-07	0x00	0x0F
SPI	disable	0.000940685362	1.99999999e-08		
SPI	enable	0.000942385362	2.00000004e-08		
SPI	result	0.000943125362	6.4e-07	0x30	0x00
SPI	result	0.000944425362	6.4e-07	0x25	0x4A
SPI	result	0.000945765362	6.4e-07	0x74	0x43
SPI	result	0.000947125362	6.4e-07	0x00	0x42
SPI	result	0.000948405362	6.4e-07	0x00	0xE4
SPI	result	0.000949745362	6.6e-07	0x00	0x0F
SPI	disable	0.000951025362	2.00000004e-08		
SPI	enable	0.000952725362	1.99999999e-08		
SPI	result	0.000953465362	6.4e-07	0x30	0x00
SPI	result	0.000954745362	6.4e-07	0x25	0x4A
SPI	result	0.000956105362	6.4e-07	0x74	0x43
SPI	result	0.000957465362	6.4e-07	0x00	0x42
SPI	result	0.000958725362	6.4e-07	0x00	0xFC
SPI	result	0.000960085362	6.4e-07	0x00	0x0F
SPI	disable	0.000961345362	1.99999999e-08

RudolphRrr, 2022-07-18:
Uff, das kam jetzt nicht ganz so raus wie erhofft.

> ne, eigentlich haben wir nur einen workaround für dieses alignment problem zwischen 12 und 16 zeichen gemacht. Ich glaube, sonst haben wir nicht viel verändert.

Nun, da ist aber an sich kein Alignment Problem. :-)

 
> Mein bauchgefühl sagt mir, dass es ein SPI kernel bug oder ein problem mit unserem spezifischen setup hier ist - ist ja schon speziell mit diesem charge controller als computer. Mein team meinte [...]

Ich kann nur bestätigen, dass es mit jedem neuen Controller immer wieder spannend ist den dazu zu bringen exakt das auf dem SPI auszuspucken was der machen soll. :-)
 
> > Die Variante die Du gewählt hast, so mit Chip-Select und Adresse für jedes Kommando, ist übrigens die langsamste überhaupt.
> 
> Geschwindigkeit ist mir im moment nicht so wichtig... Aber es gibt noch den burst modus, richtig?

Richtig, die erste Stufe ist EVE_start_cmd_burst() / EVE_end_cmd_burst() außen um die Liste herum, das bewirkt das nur einmal die Adresse geschickt wird.
Das ist aber immer noch komplett blockierend.

Die nächste Stufe ist dann mit DMA.
 
> ja, ich glaube das habe ich jetzt auch verstanden. Das braucht man nur, wenn man zum beispiel den RAM mit anderen bildern updaten will und das nicht flickern soll, oder? Für display list commands [...]

Ganz genau.

Hier ist noch das Binary für den Logic-Analyzer Trace, das ist für die Logic2 Software von Saleae.
[EVE3-50G_CMD_Text_test.zip](https://github.com/RudolphRiedel/FT800-FT813/files/9130982/EVE3-50G_CMD_Text_test.zip)

RudolphRrr, 2022-07-18:
Das hier ist mit Burst:
[EVE3-50G_CMD_Text_test_burst.zip](https://github.com/RudolphRiedel/FT800-FT813/files/9131027/EVE3-50G_CMD_Text_test_burst.zip)

Ohne das EVE_execute_cmd() am Ende sind das dann 672µs in diesem Beispiel auf dem Cortex-M0+ den ich hier gerade benutze.

Und mit DMA sind das dann nur noch 159µs:
[EVE3-50G_CMD_Text_test_DMA.zip](https://github.com/RudolphRiedel/FT800-FT813/files/9131036/EVE3-50G_CMD_Text_test_DMA.zip)

Wobei ein Teil der Zeit dafür drauf geht zu Debug-Zwecken REG_CMD_DL zu lesen.
Im Trace sieht man drei Mal Chip-Select runter gehen, das erste Mal für Busy, das zweite Mal eben für REG_CMD_DL, das dritte Mal für den DMA-Transfer.
Und die 135µs zwischen dem zweiten und dem dritten Transfer ist die Zeit die benötigt wird die Daten in den DMA-Puffer zu werfen - was eben auch nur so lange dauert weil mein Controller nur mit 48MHz läuft.
Oh ja, die erste beiden Transfers sind ohne DMA, die Zeit für einzelne Kommandos ist einfach zu kurz um dafür extra den DMA anzuwerfen und am Ende will man in aller Regel das Ergebnis ohnehin direkt haben und würde auf das Ende des DMAs warten.

Ok, das ist jetzt CMD_TEXT mit 7 bis 17 Zeichen und läuft. :-)

SirVer, 2022-08-07:
@RudolphRiedel sorry für radio silence. Ich schulde dir immer noch das "fully contained" code example und ein trace. Will ich auch noch machen, aber wird noch ein paar tage oder wochen dauern - wegen Urlaub und anderen prios.

RudolphRiedel, 2022-08-07:
Ich shifte auch gerade Prioritäten wegen Urlaub. :-)

SirVer, 2023-03-13:
I mach das mal zu, ich hab das Problem immernoch, aber grade keine Zeit um das weiter zu pushen. Noch mal vielen Dank für deine Hilfe!

closed 2023-03-13 by SirVer

RudolphRrr, 2023-03-13:
Ok, schade, das Problem war schon interessant. :-)