RudolphRiedel/FT800-FT813 #49: Sound issue after init

willemfirmlogik, 2022-09-30:
When using this GT911 with the FT813, after running the initialization of the GT911, the ft813 no longer plays any sound.

RudolphRrr, 2022-09-30:
Which module are you using exactly?
And just to confirm, you can activate sound when the EVE_HAS_GT911 define is not set?

willemfirmlogik, 2022-09-30:
Thanks for the response, 

We managed to fix our issue. The issue was in the initialization, we did not load the GT911 code at the correct time, so some of the configurations that we already did got lost after loading the new settings.
We are using a generic touch screen that now uses a GT911. This is connected to an FT813 which we use with an ASTSAMD51J20

closed 2022-09-30 by willemfirmlogik

RudolphRiedel, 2022-09-30:
So have you found my AN_336 implementation for the GT911 in EVE_init()?
This might be missing an #ifdef for the reset of the GT911 by GPIO03 - but this never was an issue so far.

>This is connected to an FT813 which we use with an ASTSAMD51J20, on a custom PCB that I designed.

Nice. I am mostly using ATSAME51J myself, I need the CAN-FD.