RudolphRiedel/FT800-FT813 #79: Hello Rudolph

eschmo, 2023-02-26:
I first want to thank you for your dedication to this toppic and the work youve share with us.

As im pretty sure that, 90% of all who want to work with that display driver will finally reach your page, i just want to share a (from my point) highly neccesary toppic. I hope you find it usefull and may integrate this information in your documentation.

I worked with eve 2,3 and now finally 4.
And to be honest i had problem all the time..Even with the help of your great work i didnt got any usefull results.
Then i got the first usefull results but still got hit with a lot of display crashes and uncommon issues.

Things i've coded worked some time and then it started to crash etc. without any changes.

I searched literally for month and replace nearly every componend until i finally dropped my whole desk and started to hard wire everything because i finally believed i had issues with wiring and contacts of breadboards etc.

But still the same problems...

Today i basically finally figured out what the rootcause is!

Just to explain my setup ive used arm processors all the time so i had mcu glockrates of >100mhz. actually i run a stm32H7 at 550MHz.
as the BT8x are clocked much below the interrupt edge detection is much slower. (and yes it just works on the int edges not on just be high or low)

so today i started to make experiments with the signal slew rate of the mcu and when i configured them to the slowest setup my issues magically disappeared.

so it is highly neccesary to configure the PD and CS slew rate as slow as possible when using an mcu which is clocked much higher (actually i run a stm32H7 at 550MHz) then the BT8x.

i could nowhere find this information or even any hint to it during my searches, so this may help other who have simmilar issues.

Best regards
Thomas

RudolphRiedel, 2023-02-26:
I believe the issue you encountered is something else entirely.
With faster rise and fall of the signals you get more reflections on the lines and my general recommendation would be to use series resistors of around 50 ohms for all lines.
But I am sure that you can't make the lines rise and fall faster than EVE could handle it, only when reflections / ringing occurs things could get ugly. 

This seems to be in the same category as providing a stable 3.3V supply for the logic as failing to do so might result in a odd issues with the touch controller.

Now you made me check the user manual and datasheet for STM32H7 as well as configuring one in STM32CubeIDE. :-)
Turns out the default value for most port pins is 00 which means OSPEED is set to Low Speed.

And checking my still defunct EVE_init_spi in EVE_target.c I am using "gpio_init.Speed = GPIO_SPEED_FREQ_HIGH;" for both CS and PD and this I should change to GPIO_SPEED_FREQ_LOW.

eschmo, 2023-02-28:
Thanks a lot for your feedback!

You may right but it was the first step to get it working anyhow. reaction to the slew rate is repeatably the same.
now i see some flickering and crashing when running for more then 24hours on 60fps. maybe this points to the terminatio
but im not sure yet how to implement a good termination in my environment. i will try as soon as possible but that needs some serious rewiring.

what issues did you experience with the logik supply? could it be a problem using a pwm supply for the logik?