RudolphRiedel/FT800-FT813 #101: Adafruit HUZZAH32 - ESP32 Feather

k-ljung, 2023-08-28:
I cannot get the library working with Adafruit HUZZAH32 - ESP32 Feather, I started from other ESP32 configurations but without success. Maybe I'm not getting the GPIO mapping correct.

I'm using the Riverdi RVT50HQBNWC00-B display and using this successfully with Teensy board.

This is the ESP32 board I try to configure, https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/pinouts

This is my environment config
```[env:ESP32]
platform = espressif32
board = featheresp32
build_unflags = -Os
build_flags =
        ${env.build_flags}
         -D EVE_CS=22
         -D EVE_PDN=17
         -D EVE_SCK=5
         -D EVE_MISO=19
         -D EVE_MOSI=18
         -O2
```

Any ideas?

RudolphRrr, 2023-08-29:
The GPIO configuration is definately a bit odd with ESP32 boards.
I always wonder if the numbers printed on the board are the GPIO pins or assigned Arduino pin numbers.
And some pins might have restricted use.

So the only way to verify would be using an oszilloscope or a logic analyzer.

k-ljung, 2023-08-29:
> I always wonder if the numbers printed on the board are the GPIO pins or assigned Arduino pin numbers.
And some pins might have restricted use.

Exactly, I've tried to Google it and get many different answers.

What about the build flag EVE_USE_ESP_IDF, should i use it for ESP32?

I cannot find where you are using the EVE_SCK, EVE_MISO, EVE_MOSI in your code?

Also, do you have a description how to identify the various pins with an oscilloscope?

RudolphRrr, 2023-08-29:
>What about the build flag EVE_USE_ESP_IDF, should i use it for ESP32?

That is only for Arduino and replaces the SPI class version with ESP-IDF code.
So that sacrifices compatibility with other devices on the SPI for DMA.

>I cannot find where you are using the EVE_SCK, EVE_MISO, EVE_MOSI in your code?

https://github.com/RudolphRiedel/FT800-FT813/blob/5.x/EVE_target.c
Line 348ff.

>Also, do you have a description how to identify the various pins with an oscilloscope?

Not in particular, let it toggle and measure. :-)

k-ljung, 2023-08-29:
I was a bit unclear, it's the EVE_Test_Arduino_PlatformIO project that I'm trying to get running.

RudolphRrr, 2023-08-29:
Ah ok, so the framework line is missing and I assumed incorrectly.
In this case this is https://github.com/RudolphRiedel/FT800-FT813/blob/5.x/EVE_cpp_target.cpp line 488ff.

Abd at least for the beginning you can leave out EVE_USE_ESP_IDF, the SPI class is not using ESP-IDF anymore and therefore single transfers are faster with the SPI class now.

And then there is this:
https://github.com/RudolphRiedel/FT800-FT813/blob/5.x/EVE_target/EVE_target_Arduino_ESP32.h