RudolphRiedel/FT800-FT813 #113: white screen on adafruit 1680 (KD50G21-40NT-A1)

markand, 2024-02-28:
Hi,

I'm trying to use my Adafruit 1680 with a [VM816C50A](https://www.mouser.fr/ProductDetail/Bridgetek/VM816C50A-N?qs=qSfuJ%252Bfl%2Fd5bjy5X77p4ZA%3D%3D) evaluation board (BT816), I get up to the screen turning on with its backlight but example in this repository for ESP32 leads to a white screen all time long. I've also tried the [official bridgetek repository](https://github.com/Bridgetek/EVE-MCU-BRT_AN_025) and very minimal own code with the same exact result. No even sign of graphic artifacts. I have no clue what do I miss.

The [KD50G21-40NT-A1](https://cdn-shop.adafruit.com/product-files/1680/Datasheet.pdf) datasheet is pretty minimal and I have no clue what should I set about timings.

I think I've stumbled across one of your post in a german forum [here](https://www.mikrocontroller.net/topic/394772) and I tried the settings you've shown:

```
#define EVE_PCLK 2
#define EVE_PCLKPOL 1
#define EVE_SWIZZLE 0
#define EVE_GEN 2
#define EVE_CSPREAD 0
#define EVE_VSYNC0 (0L)   /* Tvf Vertical Front Porch */
#define EVE_VSYNC1 (13L)   /* Tvf + Tvp Vertical Front Porch plus Vsync Pulse width */
#define EVE_VOFFSET (35L) /* Tvf + Tvp + Tvb Number of non-visible lines (in lines) */
#define EVE_VCYCLE (516L) /* Tv Total number of lines (visible and non-visible) (in lines) */
#define EVE_HSYNC0 (0L)   /* Thf Horizontal Front Porch */
#define EVE_HSYNC1 (88L)  /* Thf + Thp Horizontal Front Porch plus Hsync Pulse width */
#define EVE_HOFFSET (169L) /* Thf + Thp + Thb Length of non-visible part of line (in PCLK cycles) */
#define EVE_HCYCLE (969L) /* Th Total length of line (visible and non-visible) (in PCLKs) */
#define EVE_HSIZE (800L) /* Thd Length of visible part of line (in PCLKs) - display width */
```

I'm running out of ideas for this problem unless either my ribbon cable is damaged or anything else... Did you end up with a working screen?

RudolphRiedel, 2024-02-28:
>I think I've stumbled across one of your post in a german forum [here](https://www.mikrocontroller.net/topic/394772) and I tried the settings you've shown:

Woah, that was almost 8 years ago. :-)
But that lead me to the FT810CB_HY50HD timings I ended up with back then:

```
#define EVE_HSIZE (800L)
#define EVE_VSIZE (480L)
#define EVE_VSYNC0 (0L)
#define EVE_VSYNC1 (2L)
#define EVE_VOFFSET (13L)
#define EVE_VCYCLE (525L)
#define EVE_HSYNC0 (0L)
#define EVE_HSYNC1 (20L)
#define EVE_HOFFSET (64L)
#define EVE_HCYCLE (952L)
#define EVE_PCLK (2L)
#define EVE_PCLKPOL (1L)
#define EVE_SWIZZLE (0L)
#define EVE_CSPREAD (1L)
#define EVE_HAS_CRYSTAL
#define EVE_GEN 2
```

However, back then I also attache a few pages from a KD50G21 datasheet I was able to find.
And the timings in there are less attrocious than the ones in the datasheet Adafruit linked.
That datasheet is a good example why I do not want to try to use random panels, that industry has quite a number of clowns who manage to present the exact same data as everyone else in a way that is not consistent to what everyone else uses.

And punching the numbers in my spreadsheet I end up with this now:

```
#define EVE_HSIZE (800L)
#define EVE_VSIZE (480L)
#define EVE_VSYNC0 (13L)
#define EVE_VSYNC1 (16L)
#define EVE_VOFFSET (48L)
#define EVE_VCYCLE (525L)
#define EVE_HSYNC0 (40L)
#define EVE_HSYNC1 (88L)
#define EVE_HOFFSET 176L)
#define EVE_HCYCLE (952L)
#define EVE_PCLK (2L)
#define EVE_PCLKPOL (1L)
#define EVE_SWIZZLE (0L)
#define EVE_CSPREAD (1L)
#define EVE_HAS_CRYSTAL
#define EVE_GEN 3
```

This looks a bit odd though and according to my experience and to my spreadsheet, most of 800x480 displays use the same timings.
You could try the EVE_VM816C50AD setup from EVE_config.h:
```
#define EVE_HSIZE (800L) /* Thd Length of visible part of line (in PCLKs) - display width */
#define EVE_VSIZE (480L) /* Tvd Number of visible lines (in lines) - display height */
#define EVE_VSYNC0 (0L)   /* Tvf Vertical Front Porch */
#define EVE_VSYNC1 (3L)   /* Tvf + Tvp Vertical Front Porch plus Vsync Pulse width */
#define EVE_VOFFSET (32L) /* Tvf + Tvp + Tvb Number of non-visible lines (in lines) */
#define EVE_VCYCLE (525L) /* Tv Total number of lines (visible and non-visible) (in lines) */
#define EVE_HSYNC0 (0L)   /* Thf Horizontal Front Porch */
#define EVE_HSYNC1 (48L)  /* Thf + Thp Horizontal Front Porch plus Hsync Pulse width */
#define EVE_HOFFSET (88L) /* Thf + Thp + Thb Length of non-visible part of line (in PCLK cycles) */
#define EVE_HCYCLE (928L) /* Th Total length of line (visible and non-visible) (in PCLKs) */
#define EVE_PCLK (2L)
#define EVE_PCLKPOL (1L)
#define EVE_SWIZZLE (0L)
#define EVE_CSPREAD (0L)
#define EVE_HAS_CRYSTAL
#define EVE_GEN 3
```

>Did you end up with a working screen?

I had both the FT810CB_HY50HD and the FT811CB_HY50HD up and running - waaaay back.
And at least one of these is still blocking space in one of my drawers.

I never had a VM816C50A though.

>My main MCU is ESP32c6 using simple SPI.

And I might "need" to get me one of these, but on the controller side it should be pretty much the same as a ESP32-C3.

markand, 2024-02-28:
Thank you very much for that details, I've reattached better the ribbon cable because I melded it a bit and the contact was not good, now it works fine.

[image]

I'll play a bit with all your settings and see how it goes and if I see any differences. Other than that, do you have good LCD to recommend? I must say that this one looks really cheap and at startup there is a strange backlight issue when warming up, it's not really good eye looking...

Other than that, thanks for your help and details!

closed 2024-02-28 by markand

RudolphRiedel, 2024-02-28:
How the heck did you melt that? :-)
I would recommend removing it from the pink plastic as that stuff is conductive, even though with usually a fairly high resistance.

Recommending a display is a bit difficult, it depends on the application, the quality expected and the budget.
I very much prefer modules with capacitive touch for the sensitivity of the touch and the better surface finish.
And I like the displays with the extra glas on the front like the EVE3-50G or the RVT50HQBNWC00-B as I find it much easier to put these in a case - I can 3D print a case and glue in the display from the front.
Like this one: example_pics/EVE2-50G_ATSAMC21.jpg

markand, 2024-02-28:
> How the heck did you melt that? :-)

Cheap gas soldering iron on which you always forgot there is heat going through the side at the end of the tip 🫠
 
> Recommending a display is a bit difficult, it depends on the application, the quality expected and the budget. I very much prefer modules with capacitive touch for the sensitivity of the touch and [...]

Thank you very much, I'll have a look.