RudolphRiedel/FT800-FT813 #135: FT812 Display SPI communication

mvinay21, 2024-10-14:
Hi,

When we execute my code using SPI communication from SAM4C controller, we were able to change only display color, remaining commands of text or buttons not working burst commands not executing only command executing without burst only once.
Any solutions? 
SPI frequency 30MHz

RudolphRiedel, 2024-10-14:
Hello,

I am not familiar with the ATSAM4C series, I started with ATSAMC21 and ATSAME51 and from what I found now the ATSAM4C is an earlier series and somewhat "legacy" while still in production?
At least I can not find a family page on the Microchip homepage.

We are talking about chips like the ATSAM4C16CB-AU?

As this family is currently not directly supported, what did you do to integrate it?
Did you add your own EVE_target_ATSAM4C.h?
I would be interested in adding this to the library.

You probably did not add DMA code to EVE_target.c so far?

Do you have a logic analyzer?

>SPI frequency 30MHz

But hopefully not for the init phase?
And there usually is no need to go this high, especially not when using DMA.

A compileable example to look at that is stripped of everything else would be nice.
And I would get an inexpensive ATSAM4C board to play with - but I am not finding any.
Oh yes, I found the ATSAM4C32-EK for example but that one is priced way beyond what I am willing to spend on a toy to tinker with.

RudolphRiedel, 2024-10-14:
I just remembered, I do have a SAME70 Xplained and the SPI seems to be fairly similar.

mvinay21, 2024-10-14:
30MHz is internal clock of that controller,

except for that frequency for any other frequency it's not working, using same frequency for init,
using ATSAM4C16CB-AUR controller
Not using DMA
Did you add your own EVE_target_ATSAM4C.h? No.. using existing libraries only

RudolphRiedel, 2024-10-14:
>30MHz is internal clock of that controller

So that is not the resulting SPI clock then?

>except for that frequency for any other frequency it's not working

That is odd.

>using same frequency for init

At init, the SPI clock must not be faster than 11MHz.

I am playing with the ATSAME70 XPlained right not, but I am still stuck at lighting up the LED on PC8.
Also does not help that I can not select the EDBG as tool in the project properties, so I can flash but I can not debug.

I am up to this now:
    PMC->PMC_PCER0 = (1 << ID_PIOC);
    PIOC->PIO_PER = PIO_PC8;
    PIOC->PIO_OER = PIO_PC8;
    PIOC->PIO_PUDR = PIO_PC8;
    PIOC->PIO_OWER = PIO_PC8;
    PIOC->PIO_CODR = PIO_PC8;

And the LED still is off.
Yes, it is active low on this board.

So far I am not regretting that I stuck with the newer C21 / E51. :-)

mvinay21, 2024-10-15:
hi,
At init, i tried with internal 12mhz(divided by 6), it is powering on, but later i can't display anything if i increase the freq by selecting PLLB clock.
And in EVE_init() funtion wait_reset() is getting stucked, while reading CPURESET  it cant seem to give 0 for working status of all the engines, im getting 3,which makes coprocessor and everything in reset status.
And the below code is which i have modified from your repository as EVE_target_ATSAM4C.h

 #define  EVE_SPI       SPI0   /* the SPI port to use */

static inline void EVE_pdn_set(void)
{
   ioport_set_pin_level(PIO_PB8_IDX, IOPORT_PIN_LEVEL_LOW);  /* Power-Down low */
}

static inline void EVE_pdn_clear(void)
{
    ioport_set_pin_level(PIO_PB8_IDX, IOPORT_PIN_LEVEL_HIGH);  /* Power-Down high */
}

static inline void EVE_cs_set(void)
{
    ioport_set_pin_level(PIO_PA5_IDX, IOPORT_PIN_LEVEL_LOW);    /* cs low */
}

static inline void EVE_cs_clear(void)
{
   ioport_set_pin_level(PIO_PA5_IDX, IOPORT_PIN_LEVEL_HIGH);     /* cs high */
}

static inline void spi_transmit(uint8_t data)
{
    while (!(EVE_SPI->SPI_SR & SPI_SR_TDRE));  // wait for transmit complete
	EVE_SPI->SPI_TDR = data;
}

static inline void spi_transmit_32(uint32_t data)
{
    spi_transmit((uint8_t)(data & 0x000000ffUL));
    spi_transmit((uint8_t)(data >> 8U));
    spi_transmit((uint8_t)(data >> 16U));
    spi_transmit((uint8_t)(data >> 24U));
}

/* spi_transmit_burst() is only used for cmd-FIFO commands */
/* so it *always* has to transfer 4 bytes */
static inline void spi_transmit_burst(uint32_t data)
{
    spi_transmit_32(data);
}

static inline uint8_t spi_receive(uint8_t data)
{
	uint32_t  tx_data;
	
while (!(SPI0->SPI_SR & SPI_SR_TDRE));
SPI0->SPI_TDR = data;
	
while (!(EVE_SPI->SPI_SR & SPI_SR_RDRF));
tx_data = EVE_SPI->SPI_RDR;

return tx_data;
}

static inline uint8_t fetch_flash_byte(const uint8_t *p_data)
{
return (*p_data);
}

RudolphRiedel, 2024-10-15:
Hello,

I can not try this for myself here as my E70 behaves quite weird.
But according to Figures 33-5 and 33-7 in the SAM4C series datasheet using TDRE is probably not correct as TDRE beeing cleared only means that the data has been transferred from SPI_TDR to the shift register.
This is nice to use for back-to-back transfers but the way yout setup things now you probably end up setting CS to high before the transfer is done.
The right flag to use is probably TXEMPTY.
Or you check for TXEMPTY in EVE_cs_clear() and spi_receive(), similar to what I implemented in EVE_target_STM32H7.h.

And in spi_receive() you are probably reading data from the previous transfer as the current transfer is not done yet and the previous transfer already set RDRF.

A logic analyser trace would be most helpfull.

>At init, i tried with internal 12mhz(divided by 6),
>it is powering on, but later i can't display anything if i increase the freq by selecting PLLB clock.

Just looking briefly at the datasheet I am not even sure where the peripheral clock is controlled.
I found SCBR in SPI_CSRx though, why not change that to change the SPI clock?

mvinay21, 2024-10-15:
After adding TXEMPTY to our code, now it is working perfectly..
Thanks for your support.

RudolphRiedel, 2024-10-15:
Nice.
I still would be interested in that EVE_target_ATSAM4C.h in order to add it to the library.

mvinay21, 2024-10-15:
yeah, i attached the target file for SAM4C series.

[ATSAM_TARGET.zip](https://github.com/user-attachments/files/17377444/ATSAM_TARGET.zip)

And before EVE_init() to all over the program im running it with 30MHZ from PLLB using SCBR, other than that its not working in any frequency. and still facing some issues randomly when EVE_reset().  CPURESET readback giving 3 instead of 0, because of that sometimes the commands are not updating it seems.

RudolphRiedel, 2024-10-15:
>yeah, i attached the target file for SAM4C series.

Thank you!
But I need a little more in order to redistribute this, as you are the author, the copyright is with you.
This might be a mere formality here as you were using my template, but I still can not just claim this is mine, the amount of lines you added do not matter.

Here, I went over it, added a header with a license, added include guards, made EVE_SPI configureable and added the missing DELAY_MS().

[ATSAM_TARGET_proposal.zip](https://github.com/user-attachments/files/17378219/ATSAM_TARGET_proposal.zip)

So if you are ok with this, I'll integrate it.
Maybe I'll tweak it some more, I am really not fond of ASF.

But if you are not ok with this, especially the author and copyright reference, please edit the file accordingly.

> And before EVE_init() to all over the program im running it with 30MHZ from PLLB using SCBR,
> other than that its not working in any frequency.
> and still facing some issues randomly when EVE_reset(). [...]

I still would like to see a logic analyzer trace, from startup to first real screen and including PD.

mvinay21, 2024-10-16:
> And in spi_receive() you are probably reading data from the previous transfer as the current transfer is not done yet and the previous transfer already set RDRF

one small change in that file is that as you said, here im reading the data from previous transmission, so in SPI_recieve() function after sending dummy byte we should check for TXEMPTY flag and after that we should read.