RudolphRiedel/FT800-FT813 #140: Screen in portrait slider track still in landscape

Jwf68, 2024-11-04:
Hi Rudolph,
Working slider in landscape. 
Activates portrait: #define EVE_ROTATE				0x02 // screen in portrait
Slider tracker is still in portrait mode, the red area is the slider tracker. see picture.
[image]

Something seems wrong.

Best regards
Jonas Wahlfrid

    EVE_cmd_dl_burst(DL_TAG); // Enable tagging    
    EVE_cmd_dl_burst(DL_TAG + SETPOINT_SLIDER); // Tag following items
    EVE_cmd_slider(300, 250, 220, 25, 0, sliderPos, 1000);
    EVE_cmd_track(300, 250, 220, 25, SETPOINT_SLIDER);
    EVE_cmd_dl_burst(DL_CLEAR_TAG); //EVE_TAG_MASK(0); // Stop following items being tagged

RudolphRiedel, 2024-11-04:
Hi,

what is the configuration for the tracker?

RudolphRiedel, 2024-11-04:
Oh yes, if you are using the macro, are you sending the touch calibration values for the rotated screen?

Jwf68, 2024-11-04:
No I am using the touch calibration values for landscape.

RudolphRiedel, 2024-11-04:
That is the issue, the rotation also rotates the calibration, you either need to rotate after calibration or use a different set of calibration values.

Try this:
        touch_calibrate();
        EVE_cmd_setrotate(2);

Jwf68, 2024-11-04:
Perfekt I will try that tomorrow. Thank you.

Jwf68, 2024-11-05:
A different set of calibration values works fine :-)

Jwf68, 2024-11-05:
tff.c
#if defined (EVE_ME812A)
    #if EVE_ROTATE == 0x02
        // portrait    
        EVE_memWrite32(REG_TOUCH_TRANSFORM_A, 0xfffffd4c);
        EVE_memWrite32(REG_TOUCH_TRANSFORM_B, 0x0000897f);
        EVE_memWrite32(REG_TOUCH_TRANSFORM_C, 0xffe6c9f4);
        EVE_memWrite32(REG_TOUCH_TRANSFORM_D, 0x0000db8a);
        EVE_memWrite32(REG_TOUCH_TRANSFORM_E, 0x0000044c);
        EVE_memWrite32(REG_TOUCH_TRANSFORM_F, 0xffd82d36);
    #else
        // landscape     
        EVE_memWrite32(REG_TOUCH_TRANSFORM_A, 0x0000d7f9);
        EVE_memWrite32(REG_TOUCH_TRANSFORM_B, 0xffffff09);
        EVE_memWrite32(REG_TOUCH_TRANSFORM_C, 0xffe66cf0);
        EVE_memWrite32(REG_TOUCH_TRANSFORM_D, 0x0000026e);
        EVE_memWrite32(REG_TOUCH_TRANSFORM_E, 0xffff7c37);
        EVE_memWrite32(REG_TOUCH_TRANSFORM_F, 0x01eb8d19);
    #endif
#endif