RudolphRiedel/FT800-FT813 #143: Custom Font Not Working with Dynamically Loaded Strings on ESP32-S3 with EVE_RVT70H

mjerjawi, 2024-11-12:
### Environment:
- Hardware: ESP32-S3
- Display: EVE_RVT70H

### Issue:
I've created a SPIFFS manager to load text strings from JSON files stored in ESP32 flash memory, as I'm working with two languages. I've uploaded new fonts to the BT817 flash and loaded them using EVE_cmd_setfont2().

### Problem:
When I try to use the new fonts with strings passed directly to EVE_cmd_text(), they work perfectly
and my get_string() function works correctly with the built-in fonts.

```c
EVE_cmd_text(16, 87, 12, EVE_OPT_CENTERY, "Sensor is Unpluged");
/* This also works */
EVE_cmd_text(16, 87, 28, EVE_OPT_CENTERY, get_string("ERROR_UNPLUGED"));
```

However, when I pass the string from a SPIFFS function with the uploaded fonts , the screen displays infinite lines!

```c
EVE_cmd_text(16, 87, 12, EVE_OPT_CENTERY, get_string("ERROR_UNPLUGED"));
```

RudolphRiedel, 2024-11-12:
That is odd.
So the strings are null-terminated.
Are these JSON really UTF-8 encoded?

mjerjawi, 2024-11-12:
yes, its very odd even this not working! 
```c
    char testStr[32] = {0};
    strcpy(testStr, "test strcpy");
    EVE_cmd_text(16, 87, 12, EVE_OPT_CENTERY, testStr);
```
maybe its the font, but its a google font. I'm stuck!

mjerjawi, 2024-11-13:
Update : After many attempts, all my problems were resolved by installing the latest version of EVE Asset Builder 2.12.2. I was previously using version 2.11.0.

RudolphRrr, 2024-11-13:
That is still strange, I just opened a project I did two years ago which practically does what you are doing, except for the JSON part.
And I am fairly certain that I also converted fonts with 2.11.0 when it became available.
Well, yes, maybe something went wrong when converting the font.

mjerjawi, 2024-11-13:
Yes, I completely agree. It's very strange. I used this version before too, and I don't remember any issues.

Maybe it was because of the late hour, On the other hand, it was hard to sleep on it 😅. I should have waited until morning to open the issue.

Anyway, thank you for taking the time to address the issue.

closed 2024-11-13 by mjerjawi