RudolphRiedel/FT800-FT813 #146: Multiple Custom Fonts Causing Display Issues on EVE Screen

mjerjawi, 2024-12-31:
### Environment:
•	Hardware: ESP32-S3
•	Display: EVE_RVT70H

### Issue Description
When implementing multiple custom fonts on the EVE display, I'm encountering two issues, White screen display or Screen section repetition
These issues occur  when using more than one custom font alongside icon fonts in any given screen/view.
I have loaded 5 extended format ASTC custom fonts to the EVE flash using EAB, which are loaded using `EVE_cmd_setfont2()`. The font configuration is as follows:

**Icon Fonts**
1.	app_Icons (24px) 

- Glyph size: 18,944 bytes
- XFont size: 176 bytes
- Padding: 16 bytes
- Total size: 19,136 bytes (~18.7 KB)

2.	app_Icons (30px) 

- Glyph size: 23,680 bytes
- XFont size: 176 bytes
- Padding: 16 bytes
- Total size: 23,872 bytes (~23.3 KB)

**Text Fonts**

3.	Rubik Bold (24px) 

- Glyph size: 23,040 bytes
- XFont size: 469 bytes
- Padding: 43 bytes
- Total size: 23,552 bytes (~23 KB)

4.	Rubik Regular (20px) 

- Glyph size: 23,040 bytes
- XFont size: 469 bytes
- Padding: 43 bytes
- Total size: 23,552 bytes (~23 KB)

5.	Rubik Regular (24px) 

- Glyph size: 23,040 bytes
- XFont size: 469 bytes
- Padding: 43 bytes
- Total size: 23,552 bytes (~23 KB)

### Questions
1.	Is this a known limitation of the EVE display?
2.	Are there any configuration issues in my implementation?

RudolphRiedel, 2024-12-31:
While displaying directly from FLASH works, there is only limited bandwidth for the transfers to do so. Please avoid using the flash directly, especially using fonts requires a lot of non-sequential memory reads.
In order to use a .glyph from RAM_G you need to adjust the address within the .xfont file, 

Like this for example:
EVE_cmd_flashread(MEM_FONT_29, 142528, 312);
EVE_cmd_flashread(MEM_FONT_GLYPH_29, 108160, 34368);
EVE_memWrite32(MEM_FONT_29 + 32, MEM_FONT_GLYPH_29);

Oh yes, please note that CMD_FLASHREAD expects the size to be a multiple of 4, so "XFont size: 469 bytes" does not work directly, you would need 472 bytes.

mjerjawi, 2025-01-09:
Thank you, Rudolph.

The solution worked smoothly. It initially took some time to understand the implementation of EVE_memWrite32(), but the issue is now resolved.