RudolphRiedel/FT800-FT813 #148: Touch using the GT911 do not work

Jwf68, 2025-03-19:
Hi Rudolph,
After updating to code version with the latest commit 
”[- added more functions for display-list commands](https://github.com/RudolphRiedel/FT800-FT813/commit/c2a0159a82dfb9a92347815d663ef9e2b57e3889)
 [RudolphRiedel](https://github.com/RudolphRiedel/FT800-FT813/commits?author=RudolphRiedel) committedon Dec 16, 2024”

The touch using the GT911 do not work anymore.
In order to adapt the library to our needs I have made the changes listed in the attached files.

Why does touch do not work anymore?

Best regards

Jonas Wahlfrid

[NotesTarget.txt](https://github.com/user-attachments/files/19336859/NotesTarget.txt)

[NotesConfig.txt](https://github.com/user-attachments/files/19336870/NotesConfig.txt)

RudolphRiedel, 2025-03-19:
There was a bug some months ago that prevented GT911 from working on AVR8, but I fixed that.
I just checked, dug out an EVE2-50G which has a FT813, 800x480 and a GT911 chip, so pretty similar to your config.
I took my PlatformIO Arduino project, cleaned it, cleared the PIO cache (pio system prune) and checked that the installed library files are the latest.
Built that, put it on the UNO and touch works.

Now I grabbed my STM32F407 board and opened STM32CubeIDE, checked the project and it was configured for EVE2-50G.
I wired the Nucleo to the display and started it - works, including touch.
I updated the library files in the project, built it and it still works.
I switched to using the slow HAL_SPI functions in EVE_target_STM32.h - still works.
I disabled DMA thru the project properties - still works.
I commented out sending the calibration values and did a coldstart - still works.

I am running out of things to try.
I even just checked AN_336 which is the source of the GT911 patch for FT811/FT813, there is no extra set of data for the FT811, both the FT811 and the FT813 are using the same patch.

If this is a library issue I can not reproduce it right now.

Jwf68, 2025-03-19:
Hi Rodolfo,
When I think about it, it was something with pin CTP_RST_N and / or CTP_INT_N that needed to be changed for the FT811. 

I apologize a thousand times for this. I had forgotten that I changed the pin. I will come back with exactly what needs to be changed.

Best regards
Jonas Wahlfrid

Jwf68, 2025-03-20:
Hi Rodolfo!
Specific to FT811 together with GT911 touch controller we have to use GPIO1 to reset GT911.
Attached files describe the changes

I apologize a thousand times for this. I had forgotten that I changed the GPIO pin.

Best regards
Jonas Wahlfrid

[image]
[NotesEVE_commands.txt](https://github.com/user-attachments/files/19362626/NotesEVE_commands.txt)
[NotesConfig.txt](https://github.com/user-attachments/files/19362630/NotesConfig.txt)
[NotesTarget.txt](https://github.com/user-attachments/files/19362631/NotesTarget.txt)

RudolphRrr, 2025-03-20:
Hmm, ok, your schematic shows GPIO1 connected to INT though.
Anyways, this probably needs a couple of extra lines in EVE_commands.c, right after the patch for EVE2 + GT911 is applied.

Jwf68, 2025-03-20:
"Hmm, ok, your schematic shows GPIO1 connected to INT though."
The schematic is from https://brtchip.com/wp-content/uploads/2024/04/BRT_AN_090_EVE_Working-with-Capacitive-Touch-Screen.pdf page 8 figure 2.

"Anyways, this probably needs a couple of extra lines in EVE_commands.c, right after the patch for EVE2 + GT911 is applied."
Pseudo code on page 30 https://brtchip.com/wp-content/uploads/2024/04/BRT_AN_090_EVE_Working-with-Capacitive-Touch-Screen.pdf

RudolphRiedel, 2025-03-20:
Ok, I see you patched it, does it work for you like this?

Jwf68, 2025-03-21:
"Ok, I see you patched it, does it work for you like this?"
Yes after changing from GPIO3 to GPIO1 the touch controller works :-)